Koenkk/zigbee2mqtt #6070: Feedback on Exposes feature

flatsiedatsie, 2021-02-03:
This isn't a "problem", but I thought I'd give some snippets of feedback on the exposes functionality after having just become [intimately familiar](https://github.com/kabbi/zigbee2mqtt-adapter) with it.

1.
So far `linkquality` is the only expose I've come accross that doesnt have an underscore in its name to separate two words. I had to add some code so that it would be turned into `link quality` in the UI. 

2.
Color is a complex property, and it was a lot of work to implement because in my case this complex structure had to be turned into a HEX color code. Part of me wishes there was an option for Zigbee2MQTT to simulate such alternative properties. Similarly, it would rock if it was possible to update an `XY` color property by sending something like `"color":"FF3399"` value to the `/set`.In other words, if Zigbee2MQTT could have an internal forgiving color translation ability. But I realise this `simulated_color` is just a nice to have :-)

3.
It would be great if there was a `simulated binary` for some buttons that currently only output an `enum`.

4.
The expose API doesn't seem to distinguish types of numbers. While in my addon it's valuable to know if a value is an int, float or percentage, since these are shown in the UI differently. It's possible to work around this - and I have - by looking at the `maximum`, `multipleOf` and `unit` properties. But I'm curious if Zigbee devices don't offer this data already.

Similarly, it would be great if a percentage value could already be translated to values between 0 and 100. So far I've seen some percentage values (brightness of IKEA lights) range between 0 and 255. I was able to create some translation code for that, but it might be nice if the Zigbee2MQTT addon does this, so that all downstream software won't have to implement this individually.

Another side note: sometimes the maximum value is 254, and sometimes it's 255. That's probably due to the Zigbee devices' implementation. But assuming these are all byte values, it was interesting to see.

These are just small thoughts. I think exposes is brilliant, since it will allow plugins to be developed in a more generic manner. Thank you for creating this feature, it's highly appreciated.

Koenkk, 2021-02-03:
1. Can indeed be changed but since it somewhat breaking (will e.g. rediscover all linkquality sensor for home assistant) and its only a minor cosmetic issue it's better to leave as is.
2. Sending hex color or RGB is already supported, see https://www.zigbee2mqtt.io/devices/7146060PH.html#light, e.g. `{"color":{"hex":"#547CFF"}}`.
3. I don't understand what you mean
4. The zigbee spec differentiates between int/floats but Javascript does not.  That means a value like `1.00` will always be send as `1`.

> Another side note: sometimes the maximum value is 254, and sometimes it's 255. That's probably due to the Zigbee devices' implementation. But assuming these are all byte values, it was interesting to see.

The max brightness of a bulb is always 254, 255 is invalid (should be fixed if you saw it somewhere).

flatsiedatsie, 2021-02-03:
1. Very understandable
2. Briliant!
3. 
I have an IKEA Zigbee button which has a property that can be five different action strings, two of which are "ON" and "OFF" (the others are BRIGHTNESS_UP, BRIGHTNESS_DOWN and BRIGHTNESS_STOP). This device doesn't by itself have a `state` property. So I created some code which simulates a binary toggle properties in the UI. 
- If `ON` is received, it toggles this extra binary property to on.
- If `OFF` is received, it toggles this extra binary property to off.

This is needed for the entire thing to work optimally in the Webthings Gateway. With a binary property the entire device can  be properly categorised as an `PushButton` type, which has a better representation in the UI.

4.
I understand that the transported value itself doesn't reflect "what it's supposed to be". That's precisely why it would be useful to have clear metadata about what the `numeric` value is actually supposed to be. That way the UI can optimally reflect the device's Zigbee spec.  Anyway, this isn't important if the `multipleOf` value is always set, since that's a perfect indication of a value being int or float.

Koenkk, 2021-02-04:
3. Sounds this is very specific to the integration (e.g. Webthings). In e.g. Home Assistant this is handled differently. So in my opinion this should not be in z2m
4. 
 - @nurikk how is this currently handled in the frontend? Would such info also be useful there? 
 - If I understand correctly, the `multipleOf` not only defines int vs float but also specifies the amount of decimals to expect. This would require changing all converters that send number values (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js)

flatsiedatsie, 2021-02-04:
3.
That could very well be true. I just thought it would be nice if IKEA's button-device outputs an actual boolean button in expose. After all, that same device doesn't output a brightness property, but `simulate_brightness` fixes that. I just thought a `simulate_binary` feature would make sense in this case. When I first saw the expose data for IKEA's button, I was surprised it didn't come with a binary property. Aside from the WebThings Gateway, I could imagine that other smart home controllers would also benefit from exposing a simulated binary button, just as they benefit from having a simulated brightness. If simulated brightness didn't exist, then all these smart home controllers might attempt to code this feature themselves.

What I'm basically saying is: that `simulated_brightness` feature is awesome, and that `simulated_binary` might also be really wonderful to have. It would have been in my case.

4.
I don't know. All I know is:
- if a property is known to be a percentage, then the UI can show a slider.
- if a property is known to be an int, the UI can show a number field with [+] and [-] buttons.
- if a property is known to be a float, then a more generic input field could be shown.

and vice-versa:
- if a zigbee device expects an int value, then the smart home controller should probably not send it floats.

For both of these communication directions to work well, some meta-data about the properties is important to have. It would have been useful for my integration into WebThings Gateway, and I suspect it would also be welcome for other smart home controllers that want to integrate Zigbee2MQTT. The expose feature really lowers the barier for integrating Zigbee2MQTT. I'm just offering suggestions for how to make such integration potentially even smoother.

nurikk, 2021-02-04:
> * @nurikk how is this currently handled in the frontend? Would such info also be useful there?

I don't think frontend can benefit from this, already everything in place.

[image]

>if a zigbee device expects an int value, then the smart home controller should probably not send it floats.
this is handled by converter code, and so far no problem

flatsiedatsie, 2021-03-08:
@nurikk I'm glad it's working well for the built-in UI. But there are other systems that build on `exposes`, and I was refering to those.

closed 2021-04-17 by github-actions[bot]