Koenkk/zigbee2mqtt #6071: Problem with supported Adurosmart Eria 81809 bulb

KC-inDomus, 2021-02-03:
Got a couple of **Adurosmart Eria 81809**, both not working on ZigBee2MQTT.

[image]

As you can see, it is a 81809 model, and it seems officially supported: https://www.zigbee2mqtt.io/devices/81809_81813.html

Btw, it can't be recognized. The model number (for both bulbs) is "AD-RGBW3001":

```
Zigbee2MQTT:info  2021-02-02 18:02:47: Successfully interviewed '0x00158d00038e4fa3', device has successfully been paired
Zigbee2MQTT:warn  2021-02-02 18:02:47: Device '0x00158d00038e4fa3' with Zigbee model 'AD-RGBW3001' and manufacturer name 'AduroSmart Eria' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```
So I made a try. I found the 81809/81813 code block on devices.js:

```
    {
        zigbeeModel: ['ZLL-ExtendedColo', 'ZLL-ExtendedColor'],
        model: '81809/81813',
        vendor: 'AduroSmart',
        description: 'ERIA colors and white shades smart light bulb A19/BR30',
        extend: preset.light_onoff_brightness_colortemp_colorxy(),
        meta: {applyRedFix: true},
        endpoint: (device) => {
            return {'default': 2};
        },
    },
```
so I changed the _zigbeeModel_ key as follow:

`zigbeeModel: ['ZLL-ExtendedColo', 'ZLL-ExtendedColor', ''AD-RGBW3001'],`

with no luck. Still not working.
So I did what the unsupported guide says, added this:

``` 
    {
        zigbeeModel: ['AD-RGBW3001'],
        model: 'AD-RGBW3001',
        vendor: 'AduroSmart',
        description: 'ERIA colors and white shades smart light bulb AD-RGBW3001',
        fromZigbee: [],
        toZigbee: [],
        exposes: [],
    },
```

and that's the result on the logs:

```
Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"nd0oy-1","value":true}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"modelId":"AD-RGBW3001"}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"AD-RGBW3001"}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"AduroSmart Eria"}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"AduroSmart Eria"}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"appVersion":4}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":4}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":1}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20181012"}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20181012"}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1000-0003"}' from endpoint 1 with groupID null
No converter available for 'AD-RGBW3001' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"1000-0003"}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":269,"manufacturerCode":4653}' from endpoint 1 with groupID null
Received Zigbee message from 'landing_led', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID null
No converter available for '0x00158d0003a6ed3b' with cluster 'genLevelCtrl' and type 'attributeReport' and data '{"currentLevel":254}'

```

but I'm unable to make it work. 
The very strange thing is that btw it IS a 81809 model, so why is it behaving like this?
KC

Koenkk, 2021-02-03:
The config you posted should work, perhaps some local config error. But since I believe this config will work I added it.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

KC-inDomus, 2021-02-03:
Thanks a lot, I'll made a try.

KC-inDomus, 2021-02-03:
Loaded dev image. No change. Seems to be correctly working:

[image]

but it's not. Here's what happend when just try setting on:

```
Debug  - Received MQTT message on 'zigbee2mqtt/0x00158d0003a6ed3b/set' with data '{"state":"ON"}'
Error -  Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:183:52) at Controller.callExtensionMethod (/app/lib/controller.js:382:44) at processTicksAndRejections (internal/process/task_queues.js:93:5))
```

In "State", just {} appear.

Koenkk, 2021-02-04:
Did you discard your changes from devices.js?

KC-inDomus, 2021-02-04:
yes, I removed 

`-v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters`

from docker run command and recreated the zigbee2mqtt container using **koenkk/zigbee2mqtt:latest-dev** image.

Koenkk, 2021-02-04:
Can you force remove the device through the frontend (click red bin icon) and repair it. After that provide the debug log when trying to control the device.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

KC-inDomus, 2021-02-15:
`Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"kkgcn-1","value":true}'
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"modelId":"AD-RGBW3001"}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"AduroSmart Eria"}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"appVersion":4}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20181012"}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1000-0003"}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":31}' from endpoint 1 with groupID null
Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":500,"colorTempPhysicalMin":153}' from endpoint 1 with groupID null
Received MQTT message on 'zigbee2mqtt/0x00158d0003a6ed3b/set' with data '{"brightness":134}'`

Koenkk, 2021-02-15:
Can you also provide the herdsman debug log?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Also provide the log starting from command + 30 seconds after that.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

KC-inDomus, 2021-02-16:
```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"transaction":"28ouw-2","value":true}'
Info Zigbee: allowing new devices to join.
Info Device '0x00158d0003a6ed3b' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0003a6ed3b","ieee_address":"0x00158d0003a6ed3b"},"type":"device_joined"}'
Info Starting interview of '0x00158d0003a6ed3b'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0003a6ed3b","ieee_address":"0x00158d0003a6ed3b","status":"started"},"type":"device_interview"}'
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"modelId":"AD-RGBW3001"}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"AduroSmart Eria"}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"appVersion":4}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20181012"}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1000-0003"}' from endpoint 1 with groupID null
Info Successfully interviewed '0x00158d0003a6ed3b', device has successfully been paired
Info Device '0x00158d0003a6ed3b' is supported, identified as: AduroSmart ERIA colors and white shades smart light bulb A19/BR30 (81809/81813)
Info Configuring '0x00158d0003a6ed3b'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"ERIA colors and white shades smart light bulb A19/BR30","friendly_name":"0x00158d0003a6ed3b","model":"81809/81813","supported":true,"vendor":"AduroSmart"},"type":"pairing"}'
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":31}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorTempPhysicalMax":500,"colorTempPhysicalMin":153}' from endpoint 1 with groupID null
Info Successfully configured '0x00158d0003a6ed3b'
Debug Received Zigbee message from '0x00158d0003a6ed3b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":269,"manufacturerCode":4653}' from endpoint 1 with groupID null
```

Koenkk, 2021-02-16:
This is not the herdsman debug logging, note that it's not logged to the log files but only STDOUT.

It's also not clear that me what doesn't work exactly, I don't see any errors in the provided logs.

KC-inDomus, 2021-02-16:
Didnt' understand where to get the correct ones after launching the docker container using -e DEBUG=zigbee-herdsman*

What is not working is.. the controls. Can't turn on or off, cant set colors and so on.

closed 2021-03-27 by github-actions[bot]

KC-inDomus, 2022-02-01:
Still not working. Btw, Adurosmart Eria 81809 bulb seems supported, but imine t's still not working.