Koenkk/zigbee2mqtt #6073: TuYa TS0601_thermostat when turning on window_detection, window_detection_params sets some random values

wizu, 2021-02-03:
## What happened
When I turn on window_detection feature from Z2M interface it works ok, but it also changes window_detection_params to same random (i think) values, fo exampe:
    "window_detection": "ON",
    "window_detection_params": {
        "minutes": 234,
        "temperature": 116
    },

## What did you expect to happen
It would be great if changing window_detection wouldn't change window_detection_params

## How to reproduce it (minimal and precise)
At the "Exposes" tab enable "state Enables/disables window detection on the device"

## Debug info
Zigbee2MQTT version: 1.17.1 commit: 99274c7
Adapter hardware: CC2531
Adapter firmware version: zStack12 20190619

thiscantbeserious, 2021-02-13:
Not only does it set these parameters, when you enable it the TS0601 will become completely non-functional being stuck in scheduler settings or something.

If you manually try to adjust the temperature on the valve directly with it enabled it will display weird numbers like 95, 56, 85 ... you can't even switch settings anymore.

I guess it cant handle the above 3 digit integers?

Only thing that works is when you use force or system_mode to heat. 

Good thing is when you turn off window detection it will proceed to work again. Maybe manually adjusting the above window detection parameters will do as well ...

wizu, 2021-02-13:
@thiscantbeserious oh, and now I realize I had the same issues. It was back to normal when I reset it to factory settings. I was worried it was broken for a while ;)

wizu, 2021-03-16:
Up

wizu, 2021-04-17:
unstale

wizu, 2021-05-18:
unstale

pawol, 2021-05-18:
I confirm an issue. Probably the later ts0601 tuya firmware  has changes with window detection and its params

closed 2021-06-26 by github-actions[bot]

dedors, 2021-11-23:
This is my first Zigbee valve and it took me quite some time to figure out that it doesn't work at all when window_detection is ON and not in heat/force mode. I have default values of `"minutes": 96, "temperature": 30` and no way to change it `(No converter available for 'window_detection_params')`.
I know this issue is closed, I will leave window_detection disabled until maybe some solution comes up some day.

Moes HY368 / TS0601 / _TZE200_ckud7u2l

pawol, 2021-11-24:
Known issue but completly...pissed

thiscantbeserious, 2021-11-24:
If you're pissed about a ZigBee device behaving like this on Open-Source imagine the pain one has to go trought other 868MHz devices especially "industry" solutions from big companies not originating from China (much bigger mess, believe it or not).

Most people use Window/Door sensors for Open Window detection anyway. The temperature sensors never react fast enought or accurate anyway since logically speaking you need to detect a sharp temperature drop any usually want you would want to do is turn the heat up.

pawol, 2021-11-24:
That is the reason that issue is pissed... Nobody reads the descriptions of this bug carefully We talk that programming of window detection by zigbee2mqtt works wrong and valves are hanging after it. Manual programing works well

dedors, 2021-11-24:
That a secondary function does not work I'm completely fine with. I see the main problem here that using this switch is causing unexpected behavior and that the device to not properly function anymore. And the default/stock setting is with that enabled, which makes the first use of this very unpleasant. 

Would it be a good idea to add a comment on the device page?