Koenkk/zigbee2mqtt #6079: Implement Mqtt standard specification

slobberbone, 2021-02-04:
It would be interesting to implement in this project the specification describe here :
https://homieiot.github.io/

However, MQTT doesn’t define the structure and content of these messages and their relation. An IoT device publishes data and provides interaction possibilities but a controlling entity will need to be specifically configured to be able to interface with the device.

The Homie convention defines a standardized way of how IoT devices and services announce themselves and their data on the MQTT broker.

It is thereby a crucial aspect on top of the MQTT protocol for automatic discovery, configuration and usage of devices and services.

Some other xxx2mqtt already purpose this and I think specification for MQTT is the futur of home automation.
It would be not changing the way of this project work actually, but add another way to publish.

I'm from the open source project NextDom (home automation software) and we are working on another solution from scratch. We want to organize NextDom directly with MQTT and Node-Red (Nextdom becoming a kind of orchestrator. In this way, this standard for MQTT will be very interesting for us of course, but also for all others home automation software.

Thanks for reading and give me some feed back about that !

sjorge, 2021-02-04:
There was a similar issue before:
https://github.com/Koenkk/zigbee2mqtt/issues/3258

Koenkk, 2021-02-04:
Since we now have exposes this can be implemented indeed (exposes docs: https://www.zigbee2mqtt.io/information/exposes). The homie implementation can be created as an extension, the `homeassistant.js` can be taken as a base: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js

id628, 2021-03-13:
This would really make using zigbee2mqtt MUCH easier in openHAB.

closed 2021-04-21 by github-actions[bot]