Koenkk/zigbee2mqtt #6084: homeassistant doesn't get real color

og9gjz567, 2021-02-04:
## What happened
Hi, I'm using the latest Zigbee2mqtt Edge version and paired three rgb-spots (Müller Licht 44435) successfully. The autodiscovery added all the spots. I'm now able to turn the spots on and off. I'm able to switch the brightness, too. But if I use the color-picker, the spot changes the color, but in homeassistant the color isn't changed. In the webui of z2m I see the correct color of the spot.

## What did you expect to happen
It would be nice if the color in homeassitant would be the same :D

## How to reproduce it (minimal and precise)
ha autodiscovery, change color with picker, color of spot changes in ha not

## Debug info
Zigbee2MQTT version: 1.17.0-dev
Adapter hardware: Conbee II
Adapter firmware version: 0x26680700

[image]
[image]
[image]
[image]

pixeldoc2000, 2021-02-04:
I have the same Problem but with Osram E27 RGBW Light Bulbs. The Bulb change Color but Setting in Home Assistant is wrong. Also scene does not work.

My Gledopto LED Controller works fine.

Looks like a bug.

MQTT Message Set from HA to Osram Light:
`{"state": "ON", "color": {"x": 0.157, "y": 0.457}}`

MQTT Message Answer from Osram Light:
`{"brightness":254,"color":{"h":315,"s":0,"x":0.157,"y":0.457},"color_temp":83,"linkquality":33,"state":"ON","update": "state":"idle"},"update_available":false}`
**Note the weird broken _update_state_** This does not look right...

Maybe HA can't parse the Answer because of the broken `update_state`?

Zigbee2Mqtt 1.17.1
Home Assistant 2021.1.5
Adapter: Slaesh's CC2652RB stick
Adapter Firmware: CC2652RB_coordinator_20210120

og9gjz567, 2021-02-09:
Hi I've updated home assistant and now everything works as expected.

closed 2021-02-09 by og9gjz567