Koenkk/zigbee2mqtt #6088: New manufacturer TuYa TS0601_thermostat (_TZE200_ywdxldoj)

ASNNetworks, 2021-02-04:
Hi, this thermostat is already working with z2m, however there appears to be a new manufacturer. This one is not listed in the devices.js and I can't pair it. I own one already which does pair. 

'The manufacturer name is `_TZE200_ywdxldoj` I believe the devices are identical other than that. Can you add support for this manufacturer through edge release? Can I do anything to help?

Currently this is the only entry in the devices.js 

```
    {
        zigbeeModel: ['kud7u2l'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ckud7u2l'}],
        model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',
        whiteLabel: [{vendor: 'Moes', model: 'HY369RT'}, {vendor: 'SHOJZJ', model: '378RT'}],
        meta: {tuyaThermostatPreset: tuya.thermostatPresets, tuyaThermostatSystemMode: tuya.thermostatSystemModes3},
        ota: ota.zigbeeOTA,
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
        toZigbee: [tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force_to_mode, tz.tuya_thermostat_force, tz.tuya_thermostat_preset, tz.tuya_thermostat_away_mode,
            tz.tuya_thermostat_window_detect, tz.tuya_thermostat_schedule, tz.tuya_thermostat_week, tz.tuya_thermostat_away_preset],
        exposes: [
            e.child_lock(), e.window_detection(), e.battery_low(), e.valve_detection(), e.position(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(ea.STATE_SET)
                .withAwayMode().withPreset(['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco']),
            e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
            e.max_temperature(), e.min_temperature(), e.week(), e.away_preset_temperature()],
    },
```

877dev, 2021-02-05:
@ASNNetworks i have the same device it seems. 

I tried adding support here but had an error, fancy giving it a try?

Koenkk, 2021-02-05:
Added, please check if it works

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

877dev, 2021-02-05:
@Koenkk that seems to be working perfectly for me now, thanks so much for adding quickly!

closed 2021-02-06 by Koenkk

ASNNetworks, 2021-02-06:
Thanks, just got home: it works 😄!

Murada99, 2021-02-09:
Thanks! it works for me too !!! Just received the thermostat today.

877dev, 2021-02-10:
@Koenkk can I just quickly check something mentioned in the devices page. 

(Apologies for the formatting as I'm using my phone).

Under 'local temperature' heading it says to send the "temperature calibration" payload to get the current local temp, which works.

But under 'climate heading', it says the following, which does not work:

local_temperature: Current temperature measured on the device (in °C). To read send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"local_temperature": ""}.

Maybe I am misunderstanding..

Koenkk, 2021-02-10:
@877dev this is a mistake in the docs (it's not supported). Fixed this now.

877dev, 2021-02-10:
@Koenkk good to know, thank you.

ramelito, 2021-02-14:
When will it appear in stable branch?