Koenkk/zigbee2mqtt #6101: Networkmap error (Failed to execute LQI for router)

mabusdogma, 2021-02-05:
## What happened
There’s an error when I try to get the networkmap in the zigbee2mqtt API in Home Assistant.
Before I could visualize all the devices and the mesh.

[image]

But now I see a message when its refreshing:

`Failed to execute LQI for Router`

[image]

And I get a map with some devices apparently out of the network, and no mesh at all
And even if they appear to be offline, all devices are working just fine! So I guess they are connected in some other way, or it’s just a visualization problem.

## What did you expect to happen
Get the whole networkmap with coordinator, router and devices.

## How to reproduce it (minimal and precise)
Home Assistant -> zigbee2mqtt -> Map

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531 (coordinator and router)
Adapter firmware version: 3.0

m0wlheld, 2021-02-06:
I have the same, LQI cannot be read for a CC2531 (+ Ampflifier) device, running ptvo.info firmware.

andyshilton, 2021-02-06:
I can confirm this is an issue with the TI Launchxl-CC1352P-2 as well. The map fails to load the LQI of any devices.
Home Assistant: 2021.2.1
zigbeetomqtt: 1.17.1
Adapter: TI LAUNCHXL-CC1352P-2
Adapter firmware: CC1352P2_CC2652P_launchpad_coordinator_20210120.hex

Only addition to this is that Uniflash updated the firmware of the board before flashing it with the adapter firmware. Unfortunately I didn't grab the version and now I'm struggling to figure out what version it's on. If it is needed, please let me know and I'll hunt some more.

[image]

[image]

Koenkk, 2021-02-06:
Failed to execute LQI can happen for 2 reasons
1. The network is congested and therefore no responses are send. Try increasing the sleep between network scans, change e.g. https://github.com/Koenkk/zigbee2mqtt/blob/99274c72c26ba21347908af8238fa8dfdb4fb99d/lib/extension/networkMap.js#L215 to `await utils.sleep(5);`
2.  The device doesn't support an LQI request (I know this is the case for e.g. non-pro Gledopto devices). Cannot be solved from z2m side

andyshilton, 2021-02-07:
@Koenkk  Thanks for the information. Having left it overnight to settle, mine is responding a lot better. It does seem to be to be with network congestion.

closed 2021-03-17 by github-actions[bot]

DieNummerEins, 2021-04-27:
I have the same problem. I 

Zigbee2mqtt version: 1.18.1-1

I tried 2 CC2531-sticks, one with antenna and one without. And i tried both new firmares default and source routing from
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

this error message appears even if no end device is in the network, as you can see below
[image]

DieNummerEins, 2021-04-27:
I deinstalled and reinstalled z2m and now it works with default and sorce-routing firmware, i don't see the error message anymore. But i found out, that not every tradfri device works perfectly with the sorce-routing fw, dimmer and motion sensor don't. i have 2 cc2531-sticks, one with default and the other with source-routing fw. i only can join them to my network with the default fw. i get the button-events but the map function works only with the stick, with which i joined the device with my network. reconfiguring the device with the other stick shows me the error below.
[image]

DieNummerEins, 2021-04-27:
after the error message above i cannot join any other device. after the initial pairing message nothing happens. nothing is helping (restart z2m, unplug the stick from usb). i think, i should flash the fw and start from beginning.

DieNummerEins, 2021-04-27:
the problem seems to be the source routing fw. with the default fw i am able to get all my devices correctly mapped (with lqi). some devices (ikea tradfri) i can reconfigure, i get the error message above too, but after this reconfiguring try they are all shown correctly in the map. the other devices i can delete and rejoin again.

the default fw seems to be more stable.

now i will join all my devices (22x) and maybe it will work.

SpitFireRSA, 2021-06-05:
@DieNummerEins

> the problem seems to be the source routing fw. with the default fw i am able to get all my devices correctly mapped (with lqi). some devices (ikea tradfri) i can reconfigure, i get the error message [...]

I am experiencing the same issues reported here when using the source routing.  

Did reverting back to the default firmware prove to be more stable?

DieNummerEins, 2021-06-21:
yes, it did.

SpitFireRSA, 2021-06-21:
Thanks. I reverted and stable now.

slemik99, 2022-09-12:
> Failed to execute LQI can happen for 2 reasons
>
> 1. The network is congested and therefore no responses are send. Try increasing the sleep between network scans, change e.g. [...]

Hi,
Is it possible to solve this in HA add-on mode for z2m?
I can't reach networkMap.js on my installation (
Can it be put to settings?

eoryb, 2022-10-26:
> Hi, Is it possible to solve this in HA add-on mode for z2m? I can't reach networkMap.js on my installation ( Can it be put to settings?

My question would be exactly the same?!
How can I do this modification under HAOS installation?!