Koenkk/zigbee2mqtt #6102: Got stuck adding Moes Tuya TS0601 thermostat radiator valve

877dev, 2021-02-05:
This device is unsupported, Moes TS0601 radiator valve mode on the box HY368.

I tried to follow the guide to add support, here's what I did (I use IOTstack which is a Docker install):

1) Followed instruction to clone  https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#docker
The docker run command did run but I got an error regarding node:12 not found or similar
2) Edited device.js and added the device like this:
```
 zigbeeModel: ['kud7u2l'],
        fingerprint: [
                {modelID: 'TS0601', manufacturerName: '_TZE200_ckud7u2l'},
                {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'},

        ],
        model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',
        whiteLabel: [{vendor: 'Moes', model: 'HY369RT'}, {vendor: 'SHOJZJ', model: '378RT'}, {vendor: 'Moes', model: 'HY368'}],
        meta: {tuyaThermostatPreset: tuya.thermostatPresets, tuyaThermostatSystemMode: tuya.thermostatSystemModes3},
        ota: ota.zigbeeOTA,
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
        toZigbee: [tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force_to_mode, tz.tuya_thermostat_force, tz.tuya_thermostat_preset, tz.tuya_thermostat_away_mode,
            tz.tuya_thermostat_window_detect, tz.tuya_thermostat_schedule, tz.tuya_thermostat_week, tz.tuya_thermostat_away_preset],
        exposes: [
            e.child_lock(), e.window_detection(), e.battery_low(), e.valve_detection(), e.position(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(ea.STATE_SET)
                .withAwayMode().withPreset(['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco']),
            e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
            e.max_temperature(), e.min_temperature(), e.week(), e.away_preset_temperature()],
    },

```

Basically just added _TZE200_ywdxldoj and model HY368.

3) In docker-compose.yml added the map for the directory:

volumes:
  - /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters

4) ```docker-compose pull -d``` to recreate the container....

5) In console (PortainerCE) got NPM errors, and "Error: Cannot find module 'zigbee-herdsman-converters'"

## Information about the device + link
It's exactly the same as this one already supported:
https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html

Zigbee2mqtt console: 

```Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ywdxldoj'```
..

## data/database.db entry of the device
```
{"id":20,"type":"EndDevice","ieeeAddr":"0x847127fffe0eb922","nwkAddr":25448,"manufId":4098,"manufName":"_TZE200_ywdxldoj","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ywdxldoj","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1612533808228}
```
..

michbaeck, 2021-02-05:
might be off but if you discovered the integration https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html, why adding a new one?

877dev, 2021-02-05:
> might be off but if you discovered the integration https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html, why adding a new one?

Because it's a different version of that model, and showing as unsupported. 

 manufacturer name '_TZE200_ywdxldoj

Koenkk, 2021-02-05:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/6088

closed 2021-02-05 by Koenkk