Koenkk/zigbee2mqtt #6104: Cannot configure manuSpecificSamsungAccelerometer via UI or MQTT API (bridge/request/device/configure_reporting)

crowbarz, 2021-02-05:
## What happened
I wanted to change the reporting interval on my Samsung multipurpose sensor (model `multiv4`) for cluster `manuSpecificSamsungAccelerometer` attribute `acceleration` from the default 10s/60s/1, to stop it reporting every minute. I tried to change the reporting via both the UI and via MQTT, but both methods generate errors:

```
Zigbee2MQTT:debug 2021-02-05 15:11:17: Received MQTT message on 'zigbee/zigbee2/bridge/request/device/configure_reporting' with data '{"attribute":"acceleration","cluster":"manuSpecificSamsungAccelerometer","id":"Dining Bar Door Left Sensor/1","maximum_report_interval":3600,"minimum_report_interval":"1","reportable_change":1,"transaction":"49n8c-1"}'
Zigbee2MQTT:error 2021-02-05 15:11:21: Request 'zigbee/zigbee2/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x24fd5b000109d957/1 manuSpecificSamsungAccelerometer([{"attribute":"acceleration","minimumReportInterval":"2","maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

Changing reporting for "standard" clusters/attributes (eg. `Temperature` / `measuredValue`) for the same device works:

```
Zigbee2MQTT:debug 2021-02-05 15:13:29: Received MQTT message on 'zigbee/zigbee2/bridge/request/device/configure_reporting' with data '{"attribute":"measuredValue","cluster":"msTemperatureMeasurement","id":"Dining Bar Door Left Sensor/1","maximum_report_interval":3600,"minimum_report_interval":10,"reportable_change":100,"transaction":"49n8c-2"}'
Zigbee2MQTT:info  2021-02-05 15:13:34: Configured reporting for 'Dining Bar Door Left Sensor/1', 'msTemperatureMeasurement.measuredValue'
```

Reporting for the `acceleration` attribute is set up on initial configuration in [`devices.js` for model `multiv4`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L8048). I can change the values there to the desired reporting parameters, restart the Docker container to use the updated code, then initiate a reconfigure on the device. The updated reporting parameters then show up in the UI at this point. So it would seem that the device itself accepts changes to reporting for the `acceleration` attribute.

## What did you expect to happen
The reporting configuration changes are accepted, and the UI reflects the updated reporting configuration.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.1 (Docker)
Adapter hardware: CC2652R (zzh)
Adapter firmware version: 20210120

Database entry (after successful reporting changes to `acceleration` - even though it now appears in the reporting UI, I cannot change the value there.):
```json
{"id":65,"type":"EndDevice","ieeeAddr":"0x24fd5b000109d957","nwkAddr":47461,"manufId":4362,"manufName":"SmartThings","powerSource":"Unknown","modelId":"multiv4","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,15,32,1026,1280,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"multiv4","manufacturerName":"SmartThings","powerSource":0,"zclVersion":1,"appVersion":27,"swBuildId":"R860_#71"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00219ecc7b","zoneState":1}},"manuSpecificSamsungAccelerometer":{"attributes":{"acceleration":0,"x_axis":1033,"y_axis":-1,"z_axis":20}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2109}},"genPowerCfg":{"attributes":{"batteryVoltage":16}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00219ecc7b","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00219ecc7b","endpointID":1},{"cluster":64514,"type":"endpoint","deviceIeeeAddress":"0x00124b00219ecc7b","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":64514,"attrId":16,"minRepIntval":1,"maxRepIntval":3600,"repChange":1},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"appVersion":27,"swBuildId":"R860_#71","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1612538014048}
```

Koenkk, 2021-02-05:
Should be fixed now, please let me know.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

crowbarz, 2021-02-06:
Working now in 1.17.1-dev commit: 3687863 (Docker). Thanks @Koenkk !

closed 2021-02-06 by crowbarz