Koenkk/zigbee2mqtt #6105: Moes zigbee smart switch With neutral wire is recognized as Tuya TS0012 without neutral wire device

maltyx, 2021-02-05:
## What happened
 Moes zigbee smart switch with neutral line https://www.zigbee2mqtt.io/devices/TS0601_switch.html
 is recognized by zigbee2mqtt as Tuya TS0012 _without neutral wire device_
## What did you expect to happen
If there the way to fix the way the switch to be  recognized? 

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: Current version: 1.17.1 in HA docker
Adapter hardware: CC2531 usb 
Adapter firmware version: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'

Koenkk, 2021-02-05:
Can you share the data/dabase.db entry of this + the link where you bought it (aliexpress?)

maltyx, 2021-02-06:
{"id":9,"type":"EndDevice","ieeeAddr":"0xbc33acfffed8524f","nwkAddr":43654,"manufId":4098,"manufName":"_TYZB01_vzrytttn","powerSource":"Battery","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0012","manufacturerName":"_TYZB01_vzrytttn","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":"","appVersion":65}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938445a","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938445a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":255,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1612598754037}
{"id":10,"type":"EndDevice","ieeeAddr":"0xbc33acfffed852ac","nwkAddr":55184,"manufId":4098,"manufName":"_TYZB01_vzrytttn","powerSource":"Battery","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0012","manufacturerName":"_TYZB01_vzrytttn","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":"","appVersion":65}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938445a","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938445a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":255,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1612584783013}

https://www.aliexpress.com/item/4001245197187.html?spm=a2g0v.12010612.8148356.8.2e0942d8fUomJa

maltyx, 2021-02-06:
after some searching, found this issue: https://github.com/Koenkk/zigbee2mqtt/issues/4886#issuecomment-724732174
probably its can explain my problem.. wrong firmware on the device?

Koenkk, 2021-02-06:
This is a whitelabeled TuYa device which all share the same models. I've removed the `without neutral wire` part from the description.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-06 by Koenkk