Koenkk/zigbee2mqtt #6106: CC2531 Router Firmware

AbeltjeNL, 2021-02-05:
# Bug Report
## What happened

I flashed my CC2531 with the latest router firmware (CC2531_router_2020_09_29.zip) but shows 20180926 as active firmware.

## What did you expect to happen

That after joining the router to my network the Zigbee2MQTT add-on would show the firmware 20200929. Instead it shows 20180926 as active firmware.
Also, the picture and description is from a CC2530.

## How to reproduce it (minimal and precise)

I downloaded the latest firmware from ptvo.info to see if this would make a difference. Also with this firmware it shows as 20180926.
After that I tried all the available firmware versions I could find, flashed them, repaired the stick but every firmware shows as 20180926. There is also one Zigbee2MQTT user on Tweakers.net with the same issue I'm facing.

## Debug Info
Zigbee2MQTT version: 1.17.1
Adapter hardware coordinator: (CC2652R ZZH)
Adapter hardware router: CC2531
Adapter firmware version coordinator: 20210120
Adapter firmware version router: 20200929 but shows as 20180926

Koenkk, 2021-02-05:
@ptvoinfo could it be that the date was not updated in the newer firmwares?

AbeltjeNL, 2021-02-05:
[image]

@Koenkk, the naming and picture of the CC2531 as mentioned before.

ptvoinfo, 2021-02-06:
@Koenkk  Yes, I've forgotten to change it inside the firmware file (and for CC2530 too). Please, ignore it and use the date of the file. I'll fix this problem in the next build.

AbeltjeNL, 2021-02-06:
@ptvoinfo, thanx for the info!

closed 2021-02-06 by Koenkk