Koenkk/zigbee2mqtt #6113: HomeAssitant can't read state of switch exposed as light

ishioni, 2021-02-05:
## What happened
I have a couple of switches which i present as lights in home assistant. It worked perfectly when I set it up some time ago, but only now did I notice that Home Assistant can't read the state - eg they are always off, but the command to turn on IS sent and the switches do change state. No way to turn the off after that however
## What did you expect to happen
To be able to control them properly ;)
## How to reproduce it (minimal and precise)

Configs:
```
advanced:
  homeassistant_legacy_triggers: false
  availability_timeout: 300
  log_level: warn
  log_output:
    - console
homeassistant: true
mqtt:
  base_topic: zzh
  server: 'mqtt://mosquitto.lan'
permit_join: false
serial:
  disable_led: false
  port: /dev/zzh
devices: devices.yaml
groups: groups.yaml
frontend:
  port: 80
```
devices.yaml:
```
'0x00124b001e728db4':
  friendly_name: Kitchentop light
  retain: true
  homeassistant:
    switch:
      type: light
      object_id: light
```
And now HA always sees the state as OFF. If i remove the homeassistant: config from the device, it works as expected. The current state is represented properly on the mqtt side, so I think it may be something to do with ha autodiscovery?

## Debug info
Zigbee2MQTT version: 1.17.1

Mr-Groch, 2021-02-17:
I can confirm this. Home Assistannt restart solves this. But it allways returns after adding new devices, or other changes in z2m that is modifing autodiscovery topics

Koenkk, 2021-02-17:
Looks like a HA issue?

closed 2021-03-28 by github-actions[bot]