Koenkk/zigbee2mqtt #6120: Specified Extended PAN ID is ignored

markus-li, 2021-02-06:
This may be known, but I didn't find it reported. If unknown I'll research further but would appreciate any pointers as to where to look.
Outside of this issue with the Extended PAN ID all works as expected.

## What happened
The specified Extended PAN ID is ignored.

## What did you expect to happen
The specified Extended PAN ID to be used.

## How to reproduce it (minimal and precise)
With a zStack30x-type coordinator, set ext_pan_id to eg: [0xcd, 0xee, 0x3a, 0x00, 0x05, 0xa9, 0x91, 0x11] and enable debug logging. Start Z2M and see the output like the below:

```
Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/var/lib/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/var/lib/zigbee2mqtt/data/database.db.backup","databasePath":"/var/lib/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[205,238,58,0,5,169,145,17],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:info  2021-02-06 09:47:08: zigbee-herdsman started
Zigbee2MQTT:info  2021-02-06 09:47:08: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20201010,"transportrev":2},"type":"zStack30x"}'
Zigbee2MQTT:debug 2021-02-06 09:47:08: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b0005a990e0","panID":6755}
```

## Debug info
Zigbee2MQTT version: 1.17.1-dev (commit #3687863)
Adapter hardware: CC2538+CC2592
Adapter firmware version:
{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20201010,"transportrev":2}

Koenkk, 2021-02-06:
Indeed a known issue, similar to https://github.com/Koenkk/zigbee2mqtt/issues/4086. 

Either the ZStack firmware is bugged or we write the wrong NV item here: https://github.com/Koenkk/zigbee-herdsman/blob/cc896d117d1ffbe3fd482238071b47cc94e79db8/src/adapter/z-stack/adapter/startZnp.ts#L111

markus-li, 2021-02-07:
Thank you, I will look at this more closely then.

falmanna, 2021-02-07:
> Either the ZStack firmware is bugged or we write the wrong NV item here

I think former as I faced this same problem with openHAB Zigbee implementation as well as discussed in the referenced issue.

closed 2021-03-17 by github-actions[bot]