Koenkk/zigbee2mqtt #6123: Linkind ZL1000100-CCT-US-V1A02 - Color Temp Range

arfond, 2021-02-06:
## What happened

Since update to 1.17.1 the color temperature range available has changed (to 153 - 350). Previously in 1.17.0 it was 150 - 500, as noted in the device page:

https://www.zigbee2mqtt.io/devices/ZL1000100-CCT-US-V1A02.html

I have a device that can measure color temperature, on 1.17.0 a value of 500 was at around 6,500k, 150 at 2,700k. Now, 153 - 2,700k, and 350 = 6,100k.

When I publish {"color_temp": "warm"}, I get:

debug 2021-02-06 11:20:06: Received MQTT message on 'zigbee2mqtt/Linkind-LED-9W-Tunable-Lamp-01/set' with data '{"color_temp": "warm"}'
debug 2021-02-06 11:20:06: Publishing 'set' 'color_temp' to 'Linkind-LED-9W-Tunable-Lamp-01'
debug 2021-02-06 11:20:06: Requested color_temp 454 is higher than maximum supported 370, using maximum!
info  2021-02-06 11:20:06: MQTT publish: topic 'zigbee2mqtt/Linkind-LED-9W-Tunable-Lamp-01', payload '{"brightness":91,"color":{"x":0.459857791746717,"y":0.410598847355503},"color_temp":370,"last_seen":"2021-02-06T11:20:06+00:00","linkquality":93,"state":"ON"}'

The others appear to work ok - cool, coolest, warmest with the new ranges.

## What did you expect to happen

The new maximum value on the colour_temp variable appears to be too low to achieve the coolest color temperature.

## How to reproduce it (minimal and precise)

## Debug info

info  2021-02-06 08:23:43: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7)
info  2021-02-06 08:23:43: Starting zigbee-herdsman (0.13.59)
debug 2021-02-06 08:23:43: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":7023},"serialPort":{"path":"/dev/serial/by-id/usb-port0","rtscts":false}}'
info  2021-02-06 08:23:46: zigbee-herdsman started
info  2021-02-06 08:23:46: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200925,"transportrev":2},"type":"zStack3x0"}'
debug 2021-02-06 08:23:46: Zigbee network parameters: {"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":7023}

adapter zzh

Koenkk, 2021-02-06:
Zigbee2MQTT 1.17.1 reads the max and mini color temp from the bulb. It's seems in your case the bulb is lying about the minimum value. In `data/database` you should find the value `153` , does changing this to `150` fix the problem?

CC: @sjorge

sjorge, 2021-02-06:
The database entry would be helpful yeah.

color_temp just limits it to what the bulbs reports.
If it returns a min of 153 and you try to set it to 150, it will set it to 153 instead.

cool, neutral, warm are set to 250, 370, 454 as those are what the tradfri puck used and were withing the old assumed range of 150 - 500 for color temp. Additionally coolest and warmest will use the bulb reported minimal and maximum value.

If 454 is out of range both warm and warmest will be whatever the maximum the bulb supports.

arfond, 2021-02-06:
thanks for the responses.

here is the database file:

{"id":30,"type":"Router","ieeeAddr":"0xccccccfffe5cc487","nwkAddr":24979,"manufId":4456,"manufName":"lk","powerSource":"Mains (single phase)","modelId":"ZBT-CCTLight-D0106","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-CCTLight-D0106","manufacturerName":"lk","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1612590119283}

I did some tests, changing the database file to have a range of 50 - 600 - this confirms the range is indeed 153 - 370. My spectro gave me 2,700k at 370, and 6,200k at 153, it didn't got any cooler / higher in color temp with lower values of 100 or 50.

I'm happy to edit the device page if this helps with this new range info - let me know. Also, I can add a note how to include the bulbs / lamps - I had to google for this info.

Thanks again for the responses, and for this great project.

sjorge, 2021-02-06:
So the reported color temp range is correct?
We should probably update devices.js so the range is also included in the exposes info.

~ sjorge

arfond, 2021-02-07:
Yes @sjorge - the range is correct.

closed 2021-02-07 by arfond