Koenkk/zigbee2mqtt #6130: Problem with latest firmware devices all gone

frank3523, 2021-02-06:
I have a small zigbee network with gledopto and ikea and a philips hue motion sensor.
I also have a cc2530 as router installed. the coordinator is a CC2531 device.
After installing latest firmware all devices were gone. No way to get them back even not with source routing fw.

Also tried the zstack 3.0 fw but with no succes.

It was throwing me tis error all the time:
Error: Data request failed with error: 'No network route' (205)

Changed back the whole config from snapshot and restored the known good firmware and now it is working again.
Has anyone had the same problem and found a solution.??

Koenkk, 2021-02-06:
`Error: Data request failed with error: 'No network route' (205)` -> can usually be fixed by re-powering some devices on your network. If you switched to zstack 3 (which is a firmware I don't recommend) you need to repair all devices: https://www.zigbee2mqtt.io/information/FAQ.html#requires-repairing

frank3523, 2021-02-06:
I repowered a few devices but it did not work.
Glad I had a recent backup.
The only device that came back was my motion sensor philips hue device.

frank3523, 2021-02-07:
Can you advise me then what is the best combination to use for the cc2531 togeteher with cc2530.
And in what order to flash.

frank3523, 2021-02-13:
I gave it another try but still with no luck. Only changed the cc2531 firmware to latest source routing for coordinator.
So i switched back to latest working firmware and all was oke again. My devices are mostly gledopto with led strips and some ikea tradfi. I don't know how to solve this.

Koenkk, 2021-02-13:
Can you try repairing?

frank3523, 2021-02-13:
So finally managed to get it working with the default coordinator firmware. Also had to power off and on some devices. In the beginning it was very unresponsive. When I pushed the button half a minute later the light turned on. But after a while and switching on and off devices it seems to respond oke.

closed 2021-03-23 by github-actions[bot]