Koenkk/zigbee2mqtt #6138: Tuya TS0013 turns itself off again after 2 minutes

riyas1980, 2021-02-06:
If I connect the wired TS0013 Tuya switch (no-neutral) as described and paired it with Zigbee2Mqtt, it turns off again after 2 minutes. But if I remove it from Zigbee2MQTT, it is working fine. So there is nothing wrong with the cable connection. 

It looks like a known issue with deCONZ but for many Zigbee2MQTT users, it's working fine. It seems like a timer activated but I don't know where I can find this information.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: Conbee II

mikesrus, 2021-02-07:
same problem. 
i found posible solution

> to make it work with ZHA and have it not to turn off after 2 minutes, you need to pair it fisrt; then keep the button pressed until the light blinks green; wait until it stops blinking WITHOUT DOING NOTHING and it will not turn off auomatically. seller should warn this, either way, it works great but it does not work as a zigbee router 28 Dec 2020 13:35
[https://www.aliexpress.com/item/4000298926256.html?spm=a2g0s.12269583.0.0.7750c7d5qdAQWE](url)

i pair device and after sucess pairing press first button and hold until green light blink. afrer it device working without reboots.

I cannot understand the algorithm completely, but it worked.

riyas1980, 2021-02-07:
@mikesrus Thank you very much. Followed your method. It's working fine. Thanks a lot. As you said, it's not working as zigbee router. But I am really happy with the outcome. :-)

mikesrus, 2021-02-08:
Cool!
as I understand, version without the zero line is not a router. and does not show the power source.

Probably we need to collect more statistics about this method, and add it to the device description in the list of supported

closed 2021-03-18 by github-actions[bot]

ilbarone87, 2021-03-30:
Hello guys, I'm facing the same problem, i'm on slaesh dongle. 
If i try with your method, pairing and then press again first gang, the switch keeps on reparing and the lights keeps on turn it off every 2 mins.
Have you found any other solution to this?
thanks.

rgolab, 2021-04-13:
I had the same problem. Solution posted by @mikesrus works fine for me.

HA-Jon, 2022-04-18:
TS0013
by _TZ3000_wyhuocal
Lonsonho
Still having the RESET after 2 minutes.
I'm using Conbee 2 on Home Assistant.
How can I fix?

guillesanzs, 2024-07-27:
It worked for me.
Let me detail a little bit about the procedure I used:
- Pairing in ZHA by pressing the button (I used the left one) until the lights flash; the search can be done via the CONBEE-II or via a device that works as a zigbee router (I used one next to it).
- When pairing is complete, the lights should stop flashing. Then exit the pairing screen in ZHA.
- Once paired, the left button is pressed to start another pairing, but no longer searching in HA. When it starts flashing, it is released and just wait to stop flashing.
- Ready