Koenkk/zigbee2mqtt #6139: 3 new Climax devices

DhKDc, 2021-02-06:
All this 3 devices are working on Z2M but I don't know what else should i do to add them officially. Bellow is the information and the code in devices.js

## Information about the device + link
_SRAC-23B-ZBSR_
https://www.climax.com.tw/srac23zb.php

_WS-15ZB_
https://www.climax.com.tw/new/ws15zb.php

_CO-8ZB_
https://www.climax.com.tw/new/co8zb.php

```
    {
        zigbeeModel: ['SRACBP5_00.00.03.06TC', 'SRAC_00.00.00.16TC'],
        model: 'SRAC-23B-ZBSR',
        vendor: 'Climax',
        description: 'Smart siren',
        fromZigbee: [fz.battery],
        toZigbee: [tz.warning],
        exposes: [e.warning(), e.battery_low(), e.tamper(), e.battery()],
    },
    {
        zigbeeModel: ['WS15_00.00.00.14TC', 'WS15_00.00.00.10TC'],
        model: 'WS-15ZBS',
        vendor: 'Climax',
        description: 'Water leak sensor',
        fromZigbee: [fz.ias_water_leak_alarm_1],
        toZigbee: [],
        exposes: [e.water_leak(), e.battery_low(), e.tamper()],
    },
    {
        zigbeeModel: ['CO_00.00.00.15TC'],
        model: 'CO-8ZBS',
        vendor: 'Climax',
        description: 'Smart carbon monoxide sensor',
        fromZigbee: [fz.ias_carbon_monoxide_alarm_1, fz.battery],
        toZigbee: [],
        exposes: [e.carbon_monoxide(), e.battery_low(), e.tamper(), e.battery()],
    },
```

Koenkk, 2021-02-07:
Added, thanks.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-07 by Koenkk

johnny709009, 2022-08-17:
Is there any converter available for the Multi Sensor (motion, temperature, smoke detection, siren features) model number -MiniSD-SD_00.00.03.11TC and for radiator Thermostat- model number - TRV_00.00.03.09TC