Koenkk/zigbee2mqtt #6151: CC2538 + CC2592 'MAC no resources' (26))' and '(0x10: MEM_ERROR)'

jlaitinen81, 2021-02-07:
I have been running my CC2538 + CC2592 dongle with no issues since September and my Zigbee network has grown steadily to 80 devices.

However, since yesterday my Zigbee2Mqtt logs have been filled with MAC no resources and MEM_ERRORs. I added a few new HUE light bulbs to the setup, did I push the number of devices above a threshold? I was under the impression that this hardware could easily handle more than a hundred devices, especially as 50 of them are HUE light bulbs and can act as routers.

The whole setup is running on a Intel NUC dedicated to the Home Assistant OS and Zigbee2MQTT is run as an addon. All software is and has always been updated to the latest versions.

Koenkk, 2021-02-07:
@Hacker-CB could you help with this?

Hacker-CB, 2021-02-07:
I wi ask Egor

Hacker-CB, 2021-02-07:
@jlaitinen81 could you try new firmware
https://github.com/jethome-ru/zigbee-firmware/tree/master/ti/coordinator/cc2538_cc2592

jlaitinen81, 2021-02-07:
@Hacker-CB I bought the dongle as preflashed from Tindie as I was not experienced with flashing back then. I have sent the seller a message asking for advice on flashing it. I shutdown the system and did a power cycling on the dongle. It seems that it helped for now, but I am afraid that the errors will come back.

Hacker-CB, 2021-02-07:
@jlaitinen81 sorry, our team can support only our firmware.
We fixed several bugs i our f/w and I don’t know the behavior of the old firmware.
You can reflash your stick with usb/uart adapter like cp2102. More information is located on our github, just follow the link I provided.

jlaitinen81, 2021-02-07:
@Hacker-CB totally understand. I am just afraid that I will break something when upgrading. Can I buy another adapter with the same chipset, flash the newest firmware on that and start using that without the need of repairing all of my devices?

My issues also made me think about hardware backup. My home is pretty automated nowadays and I have the software side well backed up and I can quickly get a new NUC as well if it breaks. The weak point is my Zigbee network. Would a second spare dongle allow me to quickly deploy a new network if necessary?

Hacker-CB, 2021-02-07:
@jlaitinen81 It is possible to restore backup on the new adapter.
In this case you need to use CLEAR adapter with empty nvram.
Script for erasing nvram: https://github.com/Koenkk/zigbee2mqtt/blob/master/scripts/zStackEraseAllNvMem.js

0x3EC, 2021-02-07:
You should use the firmware suggested by @Hacker-CB. This fixes the problem with NV_MEM memory usage. Perhaps you are just having this problem.
If you haven't updated the firmware before, then I recommend buying another CC2538 stick and flashing it. This way you can always return to your old network. Unfortunately, there can be problems when changing firmware, although this rarely happens.

closed 2021-03-17 by github-actions[bot]