Koenkk/zigbee2mqtt #6156: Support for ConBee II

bluesailor45, 2021-02-07:
when running container with ConBee II stick from phoscon.de, herdsman fails with error message:
Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

this can be corrected when modifying **configuration.yaml** with:

serial:
  
  **_adapter: deconz_**

## Information about the device + link
..

## data/database.db entry of the device
..

Koenkk, 2021-02-07:
It's documented already, see last point of https://www.zigbee2mqtt.io/information/supported_adapters.html#notes

closed 2021-02-07 by Koenkk

bluesailor45, 2021-02-09:
Hallo,
thank you for your information. As I am not very familiar with docker, I do not see how I can access and modify the configuration.yaml file within a built docker environment. Therefor I thought to modify the configuration.yaml file in the repository.
For your help, I would appreciate it very much.
Thank you very much.

Veery kind regards 
Kurt