Koenkk/zigbee2mqtt #6158: cannot get ZZH stick working; ==>Error while starting zigbee-herdsman

HJ41, 2021-02-07:
Hi, requesting your help with my issue. I have spend hours now and cannot get it working, Neither with the FAQ and previous reported issues. 

I have bought a ZZH stick electrolama type CC2652RB  but cannot get zigbee2mgtt working. I have an rasberry pi4, running with Domoticz all fine and want to use the stick for new devices.

Flashing of the stick ran fine  using the  guide (https://electrolama.com/projects/zig-a-zig-ah/ )  Blink.bin works and also  Koenkk latest coordinator software flashed:  CC2652RB_coordinator_20210120.hex    

Zigbee2MQTT:  1.17.1
ZZH!: CC2652RB-2020
coordinator version: CC2652RB_coordinator_20210120.hex
USB port:  ttyUSB0

What happened:
After succesful flash and installs when starting Zigbee2mgtt  (npm start).  I am getting errors zigbee -herdsmann is failing to start

====
Zigbee2MQTT:info  2021-02-07 15:18:27: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-02-07.15-18-27' filename: log.txt
Zigbee2MQTT:info  2021-02-07 15:18:28: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7)
Zigbee2MQTT:info  2021-02-07 15:18:28: Starting zigbee-herdsman (0.13.59)
Zigbee2MQTT:error 2021-02-07 15:18:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-07 15:18:48: Failed to start zigbee
Zigbee2MQTT:error 2021-02-07 15:18:48: Exiting...
Zigbee2MQTT:error 2021-02-07 15:18:48: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

====
My configuration.yaml:
===================
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyUSB0
# advanced required for zzh
advanced:
  rtscts: false

===
My adapapter (dmesg) reports

[159492.715818] usb 1-1.4: new full-speed USB device number 8 using xhci_hcd
[159492.851922] usb 1-1.4: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[159492.851931] usb 1-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[159492.851936] usb 1-1.4: Product: USB Serial
[159492.858566] ch341 1-1.4:1.0: ch341-uart converter detected
[159492.863298] usb 1-1.4: ch341-uart converter now attached to ttyUSB0
========================

Did you read the FAQ? YES , but none of the guidance help me. I try several options, reflash, different USB port, re-install software. 

Hope one of you can help.

HJ41, 2021-02-07:
Apologies for the large text at the end. These are my config.yaml settings.

Mr-Groch, 2021-02-07:
I think you've flashed wrong firmware - zzh is CC2652R not CC2652RB

HJ41, 2021-02-07:
> I think you've flashed wrong firmware - zzh is CC2652R not CC2652RB

Hi Mr-Groch, i flash the CC2652R  coordinator software ...and it works!  

Many thanks, so stupid. Really thought I had RB version.

closed 2021-02-07 by HJ41