Koenkk/zigbee2mqtt #6167: Documented endpoint payload doesn't match deployment payload

xaque208, 2021-02-07:
## Is your feature request related to a problem? Please describe
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
When posting a message to `zigbee2mqtt/bridge/config/devices` to the payload contains `ieeeAddr` where in the documentation it mentions `ieee_address` as the key.

https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgedevices

Which is correct?  Further, if there were another method of consuming the structs in use by this project, such as protobufs or maybe this new cuelang thing, so that I could rely on the same structures in use by this project, that could be interesting.

Koenkk, 2021-02-08:
- `zigbee2mqtt/bridge/config/devices` = the legacy api, see first section of https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#mqtt-topics-and-message-structure. Here the prop is `ieeeAddr`.
- `zigbee2mqtt/bridge/devices` (the one you linked the docs of) is the new api, here the prop is `ieee_address` as documented.

closed 2021-02-08 by Koenkk

xaque208, 2021-02-13:
I see my oversight now.  Thank you for the pointer.