Koenkk/zigbee2mqtt #6168: Power-on behavior !recover

orrpan, 2021-02-07:
## Is your feature request related to a problem? Please describe
I wish to have to have a “reverse-recover” Power-on behavior, so if light is on and power is cut, it will be off when it boots and if off it will turn on. This way I could use a impulse spring and always have power on, but with all functionality of a regular button and z2m. 

If this is possible to do with z2m I don’t know, it might need to be added by manufacturers

sjorge, 2021-02-08:
I think  `toggle` is what you want, https://github.com/Koenkk/zigbee-herdsman-converters/blob/a66c232c702b2323e49700d8aae6cef6e6a32219/converters/toZigbee.js#L52

But none of my bulbs seem to support it.

flatsiedatsie, 2021-02-11:
IKEA tradfri power sockets seem to support it. At least, it's available in the exposed API. I haven't tested it.

closed 2021-03-22 by github-actions[bot]