Koenkk/zigbee2mqtt #6172: IKEA lightbulb not reporting correct state

createcandle, 2021-02-08:
## What happened

Using Zigbee2MQTT I turn `OFF` an IKEA lightbulb.
I physicially turn off the lightbulb, and turn it back on again. The lightbulb starts in it's default state: `ON`.

I query the state of the lightbulb (`{"state":""}` to `/get`. It is reported as `OFF` even though it isn't.

## What did you expect to happen

I expected the lightbulb state to reflect the actual state of the lightbulb.

## How to reproduce it (minimal and precise)

See the steps described above

## Debug info
Zigbee2MQTT version: Latest from Github
Adapter hardware: Conbee II
Adapter firmware version: `{"maintrel":0,"majorrel":38,"minorrel":74,"product":0,"revision":"0x264a0700","transportrev":0}`

Koenkk, 2021-02-08:
Can you provide the debug log of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

createcandle, 2021-02-09:
I've done some more testing, and now I can't reproduce what I described in the first post. If I query `/get` and the lightbulb is powered, I do get the correct state. That's the good news.

The bad news is that my addon doesn't know when it needs to do a `/get` to find out the new actual state.

Ideally (and perhaps even theoretically?), Zigbee2 MQTT would send some message to `/event` with `device_accounce`. In some cases I do in fact see such a message (about 20% of the time).

Once of twice I've seen a `wake_up` message.

But in 80% of cases if I switch the lightbulb on or off manually, I see no messages on `zigbee2mqtt/#` at all.

Is there some way for the addon to find out when these network changes occur? Since the lightbulbs are routers, I would expect that there must be some hint that something is up on a lower level of the stack?

Should manually switching lightbulbs on and off lead to consistent messages on `/event`? If so, then perhaps my Conbee2 is the issue.

As a side note, I've seen the reverse as well: if a lightbulb was last known to be `ON`, but is manually powered off, then Zigbee2MQTT will report the bulb as `ON` when Zigbee2MQTT starts up. Probably based on it's persistence data in `state.json`.

Koenkk, 2021-02-10:
This totally depends on what the bulb sends. When bulbs power on they usually send a device announce. Note that the availability feature can automatically do these polls for you: https://www.zigbee2mqtt.io/information/availability.html . If you only want to use it for proper state set the following in your `configuration.yaml`. 

```yaml
advanced:
  availability_timeout: 9999999
```

flatsiedatsie, 2021-02-11:
Once again I'm impressed that such a feature already exists. Very nice.

> If you only want to use it for proper state set the following in your configuration.yaml.

When I set `availability_timeout: 9999999`, I never saw any initial responses coming in.

I tried implementing it with a 10 second timeout, and that worked. What is the lowest timeout you would consider safe and sane for any size network?

**Suggestion**
During its startup, my addon accidentally got in a situation where it tried to query the state of a lightbulb that wasn't powered on.

```
Zigbee2MQTT:error 2021-02-11 12:29:22: Publish 'get' 'state' to '0xec1bbdfffea0e111' failed: 'Error: Read 0xec1bbdfffea0e111/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
Zigbee2MQTT:debug 2021-02-11 12:29:22: Error: Read 0xec1bbdfffea0e111/1 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)
at DeconzAdapter.<anonymous> (/home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:606:23)
at Generator.throw (<anonymous>)
at rejected (/home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65)
```

Would it be an idea to have such a `no response received` count as a failed internal ping, and send an (extra) availability update based on that?

Or is there a way to request Zigbee2MQTT to do an availability ping directly? Then this could be called as soon as Zigbee2MQTT is up, and find out what the network is like A.S.A.P.

**Possible bug**
I also came across the strange situation where the initial availability message for a light said that it was available, despite it being unpowered.

To reproduce:
- `availability_timeout: 10`
- stop Zigbee2MQTT
- manually power off IKEA lightbulb
- start Zigbee2MQTT

Does the availability feature use the last know data from `state.json` to fire an initial salvo of messages? If so, doesn't that defeat the purpose of the availability feature?

flatsiedatsie, 2021-02-11:
Looking a little closer, it seems that when Zigbee2MQTT shutsdown, it sends an availability `offline` message for each device. Similarly, when Zigbee2MQTT starts, it sends an `online` message for each device, regardless of the actual state.

If there a way to disable this functionality?

And more generally: if the goal of this is to allow all devices to be visually shown as unavailable in a UI if Zigbee2MQTT shuts down, shouldn't the `zigbee2mqtt/bridge/state` availability message be used as the basis for this?

Koenkk, 2021-02-11:
> Looking a little closer, it seems that when Zigbee2MQTT shutsdown, it sends an availability offline message for each device. Similarly, when Zigbee2MQTT starts, it sends an online message for each device, regardless of the actual state.

This is indeed a known issue (see #5430) and will be fixed soon since the new HA version allows multiple availability topics: https://github.com/home-assistant/core/issues/44671 . I just have to wait until enough people migrated to the latest ha version to prevent breaking.

createcandle, 2021-02-12:
Then I will patiently wait. Until then I've created some heuristics to deal with the situation.

closed 2021-02-12 by createcandle