Koenkk/zigbee2mqtt #6174: Iris il02_1 (v3) security keypad

goodwin, 2021-02-08:
Opening this issue to continue the closed one
https://github.com/Koenkk/zigbee2mqtt/issues/4673

goodwin, 2021-02-08:
Finally, received my Iris V3 keypad. So as I have a more technical background and more possibilities - let's try to bring new life into this device support.
Currently, playing with device config, converters etc - I'm getting big part of things working more or less properly, but accidentally found one very annoying (and strange for me) issue - I'm not getting ANY requests in the IAS Ace cluster from the keypad. So, for example, I can send a request from z2m to keypad to set some specific arm mode, and I can receive sensors data from the keypad into z2m. But, when pressing ARM button on the keypad - nothing is registered on the Z2M side (and yes - tried debug log for z2m intself, and tried debug mode for the herdsman - no observable activity on any level)
However, watching with the sniffer - I can definitely see IAS Ace packages sent from device to 0x0000 with proper command code and some payload
Here is Wireshark short capture during "arm" button was pressed on the keypad itself - you can see two IAS Ace messages with commands 0x07 (I suppose it is panelChange, for motion sensor activity) and 0x00 (arm command itself)
[iris1.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5945512/iris1.pcapng.zip)
Maybe, you have any idea what can be a problem (I'm testing all of this on the latest trunk z2m), or where I can look for an issue?

goodwin, 2021-02-08:
Also, found one strange thing in the herdsman log - after re-binding device, after some time, this part always appear - maybe, this is somehow related to missing incoming IAS packages?

021-02-08T10:32:38.651Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xd0cf5efffeb38410:30127,205,4)
2021-02-08T10:32:38.651Z zigbee-herdsman:controller:endpoint Read 0xd0cf5efffeb38410/1 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disa
bleRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":fa
lse}) failed (Data request failed with error: 'No network route' (205))
2021-02-08T10:32:38.651Z zigbee-herdsman:controller:device:log Interview - failed for device '0xd0cf5efffeb38410' with error 'Error: Read 0xd0cf5efffeb38410/1 ssIasZone(["
iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,
"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
at ZStackAdapter. (/Users/goodwin/wrk/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
at Generator.next ()
at fulfilled (/Users/goodwin/wrk/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)'
2021-02-08T10:32:38.652Z zigbee-herdsman:controller:error Interview failed for '0xd0cf5efffeb38410 with error 'Error: Read 0xd0cf5efffeb38410/1 ssIasZone(["iasCieAddr","zo
neState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCo
de":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

goodwin, 2021-02-08:
However, after this error I'm able to send packets to the keypad, and receive battery/temperature reports

goodwin, 2021-02-08:
From digging more - i can't catch IAS Enroll Request from the keypad, going to the Coordinator. However, z2m writes IAS CIE address correctly. Still trying to play with a different sequence of actions for triggering IAS enroll, but no luck as of now...

goodwin, 2021-02-08:
@Koenkk Now i really need some expert help from you - I've found another strange fact - all communication between keypad and coordinator is made between endpoints 1-to-1, however all events that are sent by keypad with IAS Ace packages - are sent to coordinator endoint 3. And I have no idea why and what to do with this... IAS Zone looks properly enrolled (I did read from ssIasZoneto check that ieee address and zonestate are correct), and sniffer shows packets in the air, but I have strong feeling that they are addressed to the wrong doors...

goodwin, 2021-02-09:
If that makes any difference - I'm using cc2652 with the latest "launchpad" firmware

Koenkk, 2021-02-09:
I cannot decrypt your sniff since I don't have your network key but probably the reason why zigbee-herdsman doesn't get these message is that iasAce is not in the cluster list of endpoint 3 of the coordinator.

You can add it by changing https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/startZnp.ts#L38 to: 

`{...EndpointDefaults, endpoint: 3, appprofid: 0x0105,
        appnuminclusters: 1,
        appinclusterlist: [Zcl.Utils.getCluster('ssIasAce').ID]
},`

You should be able to find the compiled `.js` variant of this file under `node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js` (don't change the .ts variant since that won't be used!).

Note that this code is only re-run on a fresh stick **without** a backup being restored. So reflash your stick and remove `data/coordinator_backup.json` .

goodwin, 2021-02-09:
Do you have any idea why it may send packages to the wrong endpoint? Can it be in some way hardcoded in the firmware of the keypad?
I'll definitely try to add ssIasAce to the 3rd endpoint and see if there'll be any difference

P.S. Forgot about encryption. Use keys 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39 and 9E:3F:42:0A:81:9C:8E:17:6C:96:C2:28:14:C3:AB:0D to decrypt it. This is test installation, this private key has no value.
P.P.S. If I can provide any other useful information (any kind of log, sniff etc. with any followed procedure) that will help to investigate that issue - just let me know

Koenkk, 2021-02-09:
> Can it be in some way hardcoded in the firmware of the keypad?

Probably yes

goodwin, 2021-02-09:
What I did:
 - changed endpoint configuration in all startZnp.js that I've found (per your recommendation)
 - reflashed stick
 - erased stick NVRAM
 - cleared z2m configuration (states, database, coordinator backup)
 - re-joined keypad
As result - same behavior - I can't see anything in z2m log or herdsman debug output
Next, I've made few changes to pass coordinator object directly to the config function, and configured things to bind ssIasAce to the coordinator endpoint 3. And - still same, nothing on the z2m side...
[image]
Also, here is small pcap while pressing "arm" button (decryption keys are same as before)
[iris3.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5951068/iris3.pcapng.zip)

goodwin, 2021-02-09:
Ha! I did it!
Zigbee2MQTT:debug 2021-02-09 14:06:01: Received Zigbee message from '0xd0cf5efffeb38410', type 'commandArm', cluster 'ssIasAce', data '{"armmode":3,"code":"","zoneid":255}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 14:06:01: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffeb38410', payload '{"action":"arm_all_zones","action_code":"","action_zone":255,"battery":100,"linkquality":144,"voltage":5900}'

goodwin, 2021-02-09:
@Koenkk another problem was that endpoint 3 on the coordinator had a profile id 0105 (CBA), but we need profile 0104 (HA). After changing profile for this endpoint, and repeating my steps - now I'm getting all the IAS messages from my keypad. So, now I can finish preparing a profile for this keypad to handle all the functions.
Another question - do we need any endpoint with 0105 profile? In my future PR i'll reconfigure endpoint 3 for 0104 and IasAce in-cluster, and we will have no other endpoint with 0105 profile (and I'm unsure if it's really used by any device)

Koenkk, 2021-02-09:
Can you check if it also works with the wildcard profile? `0xFFFF`

goodwin, 2021-02-09:
No, unfortunately, it is not working with the profile 0xFFFF. Only with 0104

HASSHUN, 2021-02-09:
Hello Both, 
I write here first before opening a new issue, cause I think my problem is quite similar to goodwin's. The device I want to configure is a Linkind Keypad. https://zigbeealliance.org/zigbee_products/linkind-keypad/

The only difference is that this device has hub and can only trigger command when it's connected. Seems like it switches endpoints just like the Iris keypad.
It speaks to my Conbee II only after resetting the keypad but even then it's quite closemouthed.

Debug: Received Zigbee message from '0x588e81fffe23364b', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID null
Info: MQTT publish: topic 'zigbee2mqtt/0x588e81fffe23364b', payload '{"battery":100,"linkquality":65,"voltage":3100}'
Debug: Received Zigbee message from '0x588e81fffe23364b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":822935558,"imageType":2308,"manufacturerCode":4456}' from endpoint 1 with groupID null

I configured it using patterns from Xfinity and 3400-D, but only get battery, linkq, and voltage.
Unfortunately, I'm not as into software engineering as Goodwin, but tell me what info you need and I'll try to provide. thy in advance!

goodwin, 2021-02-09:
> Hello Both,
> I write here first before opening a new issue, cause I think my problem is quite similar to goodwin's. The device I want to configure is a Linkind Keypad. [...]

Best what you can do in that case - pick some cc2531 stick, and sniff some traffic during interactions with a keypad. Most interesting is to catch something in the IAS Ace cluster, sent FROM device TO 0x0000 (coordinator). I don't know other way to investigate what Is going on in the air

HASSHUN, 2021-02-09:
Can it be done with ConBee II and wireshark as well?

goodwin, 2021-02-09:
if ConBee can act as sniffer (not sure, never seen conbee), and you have two dongles to work with (you need one as coordinator to make interactions with keypad, and second one to listen air traffic and write it) - then you are good to go. otherwise, you'll have to bring second device that can work as zigbee sniffer - cheapest option, I suppose, is 2531 (and it is 100% working)

HASSHUN, 2021-02-09:
Well, if none of you have experience with this Linkind Keypad and don't have any hint how to make it work without "sniffing" the data, then I guess I have to take the lead. :)
thx, anyway

I'll open then a new ticket once I have the stick ready. Is their any "For Newbies" guide to Wireshark online?

goodwin, 2021-02-09:
You may start from this article https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

HASSHUN, 2021-02-11:
Okay, I got the the Sniffer ready and working. I can identify my devices, the Linkind Keypad as well. 
Just one more question:
What is specifically what I should look for in the data traffic? In other words, how can I read these lines?
(Is there a link to another Z2M guide on this as well?)

goodwin, 2021-02-11:
Record some traffic, especially after having your keypad binded and during any event from keypad (pin code entered, mode switch triggered by arm pressed etc.) and post it somewhere. You can create another issue and reference me - I'll try to check if your issue is similar to mine or not.

serenewaffles, 2021-03-03:
I have a few of the Iris iL02_1 keypads and would like to use them with Home Assistant. Was this integration ever merged to the main branch?

jchurchward, 2021-03-06:
> I have a few of the Iris iL02_1 keypads and would like to use them with Home Assistant. Was this integration ever merged to the main branch?

I am chasing Iris Keypad (Iris iL02_1) support as well. Did you find any further information on this?

goodwin, 2021-03-25:
Support is not merged yet, unfortunately - I'm still polishing everything. Home to finish somewhere soon - due to lack of free time, unfortunately, I can't give any ETA. But as a good part - I'm planning to share my configuration example for HA as well with support for this keypad

jchurchward, 2021-03-25:
Thanks for the update @goodwin very much appreciated. I look forward to trying it out 👍

jchurchward, 2021-04-25:
Please keep this open @goodwin is still to respond once he is ready. btw @goodwin if you share what you have I know a developer who'd be keen to help with this.

jchurchward, 2021-05-23:
Please keep this open @goodwin is still to respond once he is ready.

jchurchward, 2021-06-20:
@goodwin if your happy to share your code to date I have a couple of developers ready to work on this.

closed 2021-07-29 by github-actions[bot]

Valliess, 2021-08-02:
@goodwin I hope you are ok and maybe even in vacation...
Just a small question, did you manage to make this device work ? Could you please share some code so the people in this community can lend you a hand on this to accelerate a bit ? I really need this device supported. As I've seen, in the hubitat and even the smartthings community the device already works, so it isn't impossible after all ...

Thanks in advance for your reaction.

blashman, 2021-09-06:
Looking forward this device to be implemented on z2m environment…

Gahrlaag, 2021-11-19:
I'm also wanting to get this one going. Heck, a paypal donation is on the table even..I really want this one fixed  @goodwin  :D

KameDomotics, 2024-07-27:
Hi, is there any news? Does this keypad now work with Zigbee2mqtt? Thanks

ofercohen, 2025-02-09:
Wondering the same.  Device is paired but not recognized by Z2M.