Koenkk/zigbee2mqtt #6175: TuYa TS0121_plug current and power sensors are always 0

quantummobile, 2021-02-08:
## What happened

TuYa TS0121_plug (Blitz Wolf BW-SHP-13) current and power sensors' values are always 0.

Other sensors - power outage, energy consumption, voltage work as expected and provide meaningful values.

## What did you expect to happen

I expected to get actual polled values of current and power respectively.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.1 commit: 1.17.1
Adapter hardware: CC1352P-2
Adapter firmware version: 20210120

Core Version: 2021.2.1
Supervisor Version: 2021.02.5
Operating System: Home Assistant OS 5.11

m0wlheld, 2021-02-26:
No problems with these readings of a Blitzwolf BW-SHP13. Do you get any updates from the device.

GSzabados, 2021-03-19:
@quantummobile, what is your load? Try something big, like a vacuum cleaner. You might have too small load connected.

closed 2021-04-26 by github-actions[bot]