Koenkk/zigbee2mqtt #6177: MAC transaction expired' (240) while Switching TS0011 - TuYa Smart light switch

bongiozzo, 2021-02-08:
## What happened

In network with 17 devices there are 3 Lonsonho switches. 

```
info  2021-02-01 12:52:37: 0xbc33acfffeedfbb9 (0xbc33acfffeedfbb9): TS0011 - TuYa Smart light switch - 1 gang without neutral wire (EndDevice)
info  2021-02-01 12:52:37: 0xbc33acfffeedf987 (0xbc33acfffeedf987): TS0011 - TuYa Smart light switch - 1 gang without neutral wire (EndDevice)
info  2021-02-01 12:52:37: 0xbc33acfffeedfe1f (0xbc33acfffeedfe1f): TS0011 - TuYa Smart light switch - 1 gang without neutral wire (EndDevice)
```

Their behavior in HA is unstable . Sometimes it toggles Ok, sometimes Not.

```
Feb  8 21:54:25 pi4 npm[516]: #033[31mZigbee2MQTT:error#033[39m 2021-02-08 21:54:25: Publish 'set' 'state' to '0xbc33acfffeedfe1f' failed: 'Error: Command 0xbc33acfffeedfe1f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Feb  8 21:54:25 pi4 npm[516]: #033[34mZigbee2MQTT:debug#033[39m 2021-02-08 21:54:25: Error: Command 0xbc33acfffeedfe1f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))
Feb  8 21:54:25 pi4 npm[516]:     at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
Feb  8 21:54:25 pi4 npm[516]:     at Generator.next (<anonymous>)
Feb  8 21:54:25 pi4 npm[516]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Feb  8 21:54:25 pi4 npm[516]: #033[32mZigbee2MQTT:info #033[39m 2021-02-08 21:54:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xbc33acfffeedfe1f' failed: 'Error: Command 0xbc33acfffeedfe1f/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"0xbc33acfffeedfe1f"},"type":"zigbee_publish_error"}'
```

What steps to fix such problems are recommended?

## What did you expect to happen

Switching without problems )

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.0
Adapter hardware: CC2531 with external antenna
Adapter firmware version: 20190608

Koenkk, 2021-02-09:
Can you provide the herdsman debug log of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

bongiozzo, 2021-02-14:
> Can you provide the herdsman debug log of this?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Yes, sure! In 1-2 days.
BTW, does it have reason to upgrade adapter's firmware?

Koenkk, 2021-02-14:
I don't expect a firmware update to fix it

Please also attach your networkmap in both the working and non-working situation.

bongiozzo, 2021-02-15:
0xbc33acfffeedfe1f - has a permanent problem
0xbc33acfffeedf98 & 0xbc33acfffeedfbb9 switches with some delay

[image]

Koenkk, 2021-02-17:
- After repairing, does the device work correctly?
- I don't see the error in the provided log file, also the logging does not include the herdsman debug logging (which is only logged to STDOUT, not the log files!)

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

bongiozzo, 2021-02-20:
I removed and added this switch again. Seems that it solved the problem...
Will come back if repeated.

closed 2021-02-20 by bongiozzo

bongiozzo, 2021-02-20:
Hmm. Other ZigBee devices (Xiaomi Plug) have problems with on/off now :/

Is there some general recommendation to build more stable ZigBee network?

Koenkk, 2021-02-21:
> Is there some general recommendation to build more stable ZigBee network?

yes: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html, especially: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html#reduce-wi-fi-interference-by-changing-the-zigbee-channel

mpashka, 2021-04-30:
I have exactly the same problem. I have cc2531 with CC2531_DEFAULT_20201127 firmware. I have 14 zigbee devices (primarily xiaomi temperature sensors, motion detectors, leak detectors). And 2 switches:
Tuya 2 gang no neutral switch (TS0012)
Tuya dimmer (TS0601_dimmer).
All three (stick, switch, dimmer) are located more or less on one single line. Distance between stick and and switch is 1.8 meters, switch and dimmer - 2.5 meters.
Right after pairing device map looks like a star with stick in the center. While checking device map during the problem switch is connected to both dimmer and stick.
Device map: [device_map.graphviz.txt](https://github.com/Koenkk/zigbee2mqtt/files/6407803/device_map.graphviz.txt)
[image]
[image]

I tried to repair switch and dimmer. Problem reappears in few days. I replaced stick - the same thing.

Now I will try to disable using dimmer as a Router and switch it to act as EndDevice. Will also collect and attach herdsman debug logs.

alexanderwwagner, 2021-12-03:
Are here any updates to this topic available?
I have the same problem with my radiator thermostat from essential.
https://www.zigbee2mqtt.io/devices/GS361A-H04.html

At the beginning all works fine. But sporadically after two/three weeks the thermostat is working in read-only mode...
To workaround this happening I have to reconnect my essential thermostats...

Thank you and a nice weekend!

eraycetinay, 2021-12-20:
I have the same problem with BRT-100 
https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html
At least 1 of 3 trv is giving the same error but not the same trv always.  A different one each time. 
I just bought ac switch to use as a router to understand if it is about range and im waiting the delivery of the switch now.
If that fixes the problem i will give you an update.