Koenkk/zigbee2mqtt #6184: Groups not working

Gibbz, 2021-02-09:
## What happened
Groups are not working. The command is not being sent (or group command are not logged?)

## What did you expect to happen
To see the publish command to the group display in the log, and the devices to active.

## How to reproduce it (minimal and precise)
Create a group, add a few devices. In my case named 'chimes'
Use publish 'zigbee2mqtt/chimes/set' and {"alarm":true}
Notice that there is no log of this command, and the device is not activated.

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: zzh!
Adapter firmware version:

m0wlheld, 2021-02-09:
Just a note: Group of lights (light bulbs) do work (state, brightness).

Gibbz, 2021-02-09:
> Just a note: Group of lights (light bulbs) do work (state, brightness).

Hrmm... I wonder then if it may be related to my other issue I'm having here: #5970  
The devices in the group are the Neo NAS-AB02B0

Armageddit, 2021-03-11:
i have a problem!
if i turn on 1 bulb the group on -> MQTT say state: ON
and after view milisec all bulbs in the group show state: ON
but the other bulbs are still off!
why they say ON in MQTT??

MrSpoocy, 2021-04-16:
I agree, same problem. Using the deconz ConBee II (USB) and the Groups added. If I publish to group, nothing happen. But the mqtt say that the state has changed. If I control the devies allown, it works.

closed 2021-04-24 by github-actions[bot]