Koenkk/zigbee2mqtt #6188: Availability configuration per device / availability "ping" count

bkupidura, 2021-02-09:
## Is your feature request related to a problem? Please describe
Currently you are able to provide just a global availability (for every router in network). It would be nice if we can specify different availability for different devices (as part of `availability_passlist`).

Also it would be beneficial if we can provide count/number of "pings" which should be failed before marking device as offline.

## Describe the solution you'd like
```
availability_timeout: 3600
availability_count: 1
availability_passlist:
  - name: router1
     timeout: 600
  - name: router2
     count: 2
  - name: router3
     timeout: 120
     count: 3
  - name: router4
```

Router1 should be marked as offline after 1 failed ping which will be send every 600s.
Router2 should be marked as offline after 2 failed pings which will be send every 3600s.
Router3 should be marked as offline after 3 failed pings which will be send every 120s.
Router4 should be marked as offline after 1 failed ping which will be send every 3600s.

## Describe alternatives you've considered
None

## Additional context
In my network some devices are more important than others, so i would like to check if they are available more often. But on other hand i want to limit false positives (1 failed ping not necessary means that device is dead), so it would be nice to provide way that user can decide about number of failed availability pings.

Koenkk, 2021-02-10:
Thanks for your feature request. I'm already planning to refactor the availability feature. I will try to come up with a proposal for this soon in a new issue. Will link it here once i've created it.

Koenkk, 2021-02-14:
Incorporated this feature request in https://github.com/Koenkk/zigbee2mqtt/issues/6281 , continuing there

closed 2021-02-14 by Koenkk