Koenkk/zigbee2mqtt #6190: TuYa TS0011 not acting as an router?

MORD0R, 2021-02-09:


Koenkk, 2021-02-09:
As can be seen in the screenshot the device advertises itself as an EndDevice. This is very common for switches which have no neutral wire.

MORD0R, 2021-02-09:
> 
> 
> As can be seen in the screenshot the device advertises itself as an EndDevice. This is very common for switches which have no neutral wire. Since it is an EndDevice it does not provide router functionallity (for that you probably need the variant with neutral wire)
i think your right. Only the power source is wrong then.
https://community.home-assistant.io/t/switch-tuya-zigbee-no-neutral-wire-models-x711-x712-x713/198224

Koenkk, 2021-02-09:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-09 by Koenkk

xav-aaa, 2021-12-01:
> Sorry to ask, I have got the optional neutral wire version. I have attached the neutral wire, yet the device is still displayed as EndDevice. Is this normal or should it be possible to make this then work as a router?

I've got the same issue. I have 3 of these switches, all connected with neutral line but none of them is recognized as a router. I've the last version of zigbee2mqtt installed (v 1.22.0-3). Any idea of what to do? Thank you.

oztourer, 2022-02-09:
> Sorry to ask, I have got the optional neutral wire version. I have attached the neutral wire, yet the device is still displayed as EndDevice. Is this normal or should it be possible to make this then work as a router?

Same issue here, with TS0011, TS0012 and TS0014. All powered with live and neutral. Tested on the stable version of zigbee2mqtt and on Edge but always seen as EndDevice.

doutorinfamous, 2022-04-22:
Same problem here. 6 TS0012 with neutral wire connected, but recognized as end device.

digitui, 2022-05-31:
IS this still the case that is can not be a router as was going to install more of them, just comes up as an end device

JoryHogeveen, 2022-07-11:
Moes actually replied to me on this issue saying that none of their devices are routera at all, even the ones requiring a null line.

JoryHogeveen, 2022-07-14:
Btw, related to this issue I started the following topic on the community: https://community.home-assistant.io/t/user-experience-request-differernt-zigbee-tuya-based-touch-light-switches/439249

eskrasic, 2022-09-26:
> Moes actually replied to me on this issue saying that none of their devices are routers at all, even the ones requiring a null line.

Thanks for confirming it! I installed these switxhes everywhere in my house and try to find a solution for a few monthes. It seems like need to install other zigbee routing device to cover the full area.

bleughb, 2023-05-15:
Heh, Just found this page, I have a bunch of Mercator IKUU switch devices and all of them only show as end devices!, D'oh!

bleughb, 2023-06-09:
My solution was to purchase a bunch of cheap zigbee power monitoring passthrough plugs/sockets. they're usually <$15 AUD a pop from the regular sites and therfore better value than a similarly priced / more expensive dedicated router / repeater!

oztourer, 2023-06-09:
> My solution was to purchase a bunch of cheap zigbee power monitoring passthrough plugs/sockets. they're usually <$15 AUD a pop from the regular sites and therfore better value than a similarly priced / more expensive dedicated router / repeater!

That's a good idea. Not all include routing - what brand did you use?

bleughb, 2023-06-09:
> 

https://www.aliexpress.com/item/1005004448388719.html?spm=a2g0o.order_list.order_list_main.81.58d91802TGfzaj

AUBEES tuya zigbee - 

[image]

acca84, 2023-11-13:
Hello, i have two of these plugs, both are recognized as router, but no device connects to those routers.
Do i have to wait ? I have paired the routers one hour ago 
Thanks