Koenkk/zigbee2mqtt #6191: Can't configure manual reporting on Tradfri bulb not added to group bound to remote

crowbarz, 2021-02-09:
## What happened
Manually configured reporting settings for my Tradfri bulbs (eg. genOnOff, LevelCtrl) don't work when the bulb is not added to a group that is bound to a remote.

## What did you expect to happen
Manually configured reporting should also enable reports to be generated by the bulb, whether or not the bulb is a member of a group that is bound to a remote.

## How to reproduce it (minimal and precise)
NOTE: deprecated reporting feature has been turned off.

1. Have Tradfri bulb (`Attic Stairs Ceiling 2`, model `TRADFRI bulb GU10 W 400lm`) joined onto network, and a group (#301 `Attic Stairs Ceiling Lights`) bound to a Tradfri 5-button remote (model `TRADFRI remote control`). Bulb is _not_ added to this group, and no reporting is configured on the bulb.
1. (09:56:37) Configure reporting for `genOnOff` and `genLevelCtrl` clusters on the bulb for max interval 32s, then wait 32s. No reports are received.
```
Zigbee2MQTT:info  2021-02-09 09:56:37: Configured reporting for 'Attic Stairs Ceiling 2/1', 'genOnOff.onOff'
Zigbee2MQTT:info  2021-02-09 09:56:37: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"onOff","cluster":"genOnOff","id":"Attic Stairs Ceiling 2/1","maximum_report_interval":"32","minimum_report_interval":"0","reportable_change":0},"status":"ok","transaction":"eztva-47"}'
Zigbee2MQTT:info  2021-02-09 09:56:39: Configured reporting for 'Attic Stairs Ceiling 2/1', 'genLevelCtrl.currentLevel'
Zigbee2MQTT:info  2021-02-09 09:56:39: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"currentLevel","cluster":"genLevelCtrl","id":"Attic Stairs Ceiling 2/1","maximum_report_interval":"32","minimum_report_interval":"3","reportable_change":"1"},"status":"ok","transaction":"eztva-48"}'
```
3. (10:00:34) Add bulb to group. Logs report reporting successfully set up for `genOnOff` and `genLevelCtrl` clusters. UI shows the default reporting (with max interval 3600s) has been configured on bulb, and that Coordinator now has bindings to `OnOff` and `LevelCtrl` clusters. 
```
Zigbee2MQTT:info  2021-02-09 10:00:34: Adding 'Attic Stairs Ceiling 2' to 'Attic Stairs Ceiling Lights'
Zigbee2MQTT:info  2021-02-09 10:00:35: MQTT publish: topic 'zigbee/zigbee1/bridge/log', payload '{"message":{"friendly_name":"Attic Stairs Ceiling 2","group":"Attic Stairs Ceiling Lights"},"type":"device_group_add"}'
Zigbee2MQTT:info  2021-02-09 10:00:35: MQTT publish: topic 'zigbee/zigbee1/bridge/response/group/members/add', payload '{"data":{"device":"0x000b57fffeb2c88e/1","group":"Attic Stairs Ceiling Lights"},"status":"ok"}'
Zigbee2MQTT:info  2021-02-09 10:00:35: Succesfully setup reporting for 'Attic Stairs Ceiling 2/1' cluster 'genOnOff'
Zigbee2MQTT:debug 2021-02-09 10:00:35: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 301
Zigbee2MQTT:info  2021-02-09 10:00:35: Succesfully setup reporting for 'Attic Stairs Ceiling 2/1' cluster 'genLevelCtrl'
Zigbee2MQTT:debug 2021-02-09 10:00:36: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'read', cluster 'genLevelCtrl', data '["currentLevel"]' from endpoint 1 with groupID 301
Zigbee2MQTT:debug 2021-02-09 10:00:37: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 10:00:37: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling 2', payload '{"brightness":32,"last_seen":"2021-02-09T10:00:37.040Z","linkquality":108,"state":"OFF","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2021-02-09 10:00:37: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling Lights', payload '{"brightness":32,"state":"OFF"}'
```
4. (10:02:53) Reconfigure reporting manually to max interval 32s, then wait. Bulb starts sending reports every ~32s. (10:03:25 -- 10:05:35)
```
Zigbee2MQTT:info  2021-02-09 10:02:53: Configured reporting for 'Attic Stairs Ceiling 2/1', 'genOnOff.onOff'
Zigbee2MQTT:info  2021-02-09 10:02:53: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"onOff","cluster":"genOnOff","id":"Attic Stairs Ceiling 2/1","maximum_report_interval":"32","minimum_report_interval":0,"reportable_change":0},"status":"ok","transaction":"eztva-50"}'
Zigbee2MQTT:debug 2021-02-09 10:02:53: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 10:02:53: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling 2', payload '{"brightness":32,"last_seen":"2021-02-09T10:02:53.626Z","linkquality":111,"state":"OFF","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2021-02-09 10:02:55: Configured reporting for 'Attic Stairs Ceiling 2/1', 'genLevelCtrl.currentLevel'
Zigbee2MQTT:info  2021-02-09 10:02:55: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"currentLevel","cluster":"genLevelCtrl","id":"Attic Stairs Ceiling 2/1","maximum_report_interval":"32","minimum_report_interval":5,"reportable_change":1},"status":"ok","transaction":"eztva-51"}'
Zigbee2MQTT:debug 2021-02-09 10:02:55: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 10:02:55: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling 2', payload '{"brightness":1,"last_seen":"2021-02-09T10:02:55.545Z","linkquality":111,"state":"OFF","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2021-02-09 10:02:55: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling Lights', payload '{"brightness":1,"state":"OFF"}'
Zigbee2MQTT:debug 2021-02-09 10:03:25: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 10:03:25: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling 2', payload '{"brightness":1,"last_seen":"2021-02-09T10:03:25.436Z","linkquality":108,"state":"OFF","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:debug 2021-02-09 10:03:27: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 10:03:27: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling 2', payload '{"brightness":1,"last_seen":"2021-02-09T10:03:27.367Z","linkquality":108,"state":"OFF","update":{"state":"available"},"update_available":true}'
[...]
Zigbee2MQTT:debug 2021-02-09 10:05:35: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 10:05:35: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling 2', payload '{"brightness":1,"last_seen":"2021-02-09T10:05:35.371Z","linkquality":108,"state":"OFF","update":{"state":"available"},"update_available":true}'
```
5. (10:05:50) Remove bulb from group. Reporting is disabled and existing reporting configuration is removed.
```
Zigbee2MQTT:info  2021-02-09 10:05:50: Removing 'Attic Stairs Ceiling 2' from 'Attic Stairs Ceiling Lights'
Zigbee2MQTT:info  2021-02-09 10:05:50: MQTT publish: topic 'zigbee/zigbee1/bridge/log', payload '{"message":{"friendly_name":"Attic Stairs Ceiling 2","group":"Attic Stairs Ceiling Lights"},"type":"device_group_remove"}'
Zigbee2MQTT:info  2021-02-09 10:05:50: MQTT publish: topic 'zigbee/zigbee1/bridge/response/group/members/remove', payload '{"data":{"device":"Attic Stairs Ceiling 2","group":"Attic Stairs Ceiling Lights"},"status":"ok"}'
Zigbee2MQTT:debug 2021-02-09 10:05:50: Received Zigbee message from 'Attic Stairs Ceiling 2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-02-09 10:05:50: MQTT publish: topic 'zigbee/zigbee1/Attic Stairs Ceiling 2', payload '{"brightness":1,"last_seen":"2021-02-09T10:05:50.645Z","linkquality":111,"state":"ON","update":{"state":"available"},"update_available":true}'
[... cut ...]
Zigbee2MQTT:info  2021-02-09 10:06:49: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"currentLevel","cluster":"genLevelCtrl","id":"Attic Stairs Ceiling 2/1","maximum_report_interval":"32","minimum_report_interval":"1","reportable_change":"1"},"status":"ok","transaction":"eztva-55"}'
```

## Debug info
Zigbee2MQTT version: 1.17.1-dev commit: 3687863 (Docker)
Adapter hardware: CC2652 (zzh)
Adapter firmware version: zStack3x0 20210120

Koenkk, 2021-02-09:
> Manually configured reporting should also enable reports to be generated by the bulb, whether or not the bulb is a member of a group that is bound to a remote.

Doesn't this work correctly in step 2 (but just not in 6?)

Before step 6 can you bind the level and onoff cluster from the bulb to the coordinator?

crowbarz, 2021-02-09:
Reporting doesn't work after step 2 (after bulb first joined) or 6 (after being removed from the group). Only after the bulb is added to the group that is bound to a remote, which triggers reporting (and the binding) to be set up, does the bulb start actually reporting.

I am unable to bind from the Coordinator to the bulb or vice versa, with or without reporting already configured, when the bulb not added to the group. Can a bind to/from the Coordinator be triggered via the API?
```
Zigbee2MQTT:debug 2021-02-09 20:10:03: Received MQTT message on 'zigbee/zigbee1/bridge/request/device/bind' with data '{"from":"Attic Stairs Ceiling 2","to":"Coordinator","clusters":["genOnOff"]}'
Zigbee2MQTT:error 2021-02-09 20:10:03: Nothing to bind from 'Attic Stairs Ceiling 2' to 'Coordinator'
Zigbee2MQTT:info  2021-02-09 20:10:03: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Attic Stairs Ceiling 2","to":"Coordinator"},"error":"Nothing to bind","status":"error"}'
Zigbee2MQTT:debug 2021-02-09 20:10:45: Received MQTT message on 'zigbee/zigbee1/bridge/request/device/bind' with data '{"from":"Coordinator","to":"Attic Stairs Ceiling 2","clusters":["genOnOff"]}'
Zigbee2MQTT:error 2021-02-09 20:10:45: Nothing to bind from 'Coordinator' to 'Attic Stairs Ceiling 2'
Zigbee2MQTT:info  2021-02-09 20:10:45: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Coordinator","to":"Attic Stairs Ceiling 2"},"error":"Nothing to bind","status":"error"}'

Zigbee2MQTT:debug 2021-02-09 20:11:24: Received MQTT message on 'zigbee/zigbee1/bridge/request/device/configure_reporting' with data '{"attribute":"onOff","cluster":"genOnOff","id":"Attic Stairs Ceiling 2/1","maximum_report_interval":3600,"minimum_report_interval":60,"reportable_change":0,"transaction":"r65qd-1"}'
Zigbee2MQTT:info  2021-02-09 20:11:24: Configured reporting for 'Attic Stairs Ceiling 2/1', 'genOnOff.onOff'
Zigbee2MQTT:info  2021-02-09 20:11:24: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/configure_reporting', payload '{"data":{"attribute":"onOff","cluster":"genOnOff","id":"Attic Stairs Ceiling 2/1","maximum_report_interval":3600,"minimum_report_interval":60,"reportable_change":0},"status":"ok","transaction":"r65qd-1"}'

Zigbee2MQTT:debug 2021-02-09 20:11:32: Received MQTT message on 'zigbee/zigbee1/bridge/request/device/bind' with data '{"from":"Attic Stairs Ceiling 2","to":"Coordinator","clusters":["genOnOff"]}'
Zigbee2MQTT:error 2021-02-09 20:11:32: Nothing to bind from 'Attic Stairs Ceiling 2' to 'Coordinator'
Zigbee2MQTT:info  2021-02-09 20:11:32: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Attic Stairs Ceiling 2","to":"Coordinator"},"error":"Nothing to bind","status":"error"}'
Zigbee2MQTT:debug 2021-02-09 20:11:42: Received MQTT message on 'zigbee/zigbee1/bridge/request/device/bind' with data '{"from":"Coordinator","to":"Attic Stairs Ceiling 2","clusters":["genOnOff"]}'
Zigbee2MQTT:error 2021-02-09 20:11:42: Nothing to bind from 'Coordinator' to 'Attic Stairs Ceiling 2'
Zigbee2MQTT:info  2021-02-09 20:11:42: MQTT publish: topic 'zigbee/zigbee1/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":[],"from":"Coordinator","to":"Attic Stairs Ceiling 2"},"error":"Nothing to bind","status":"error"}'
```

crowbarz, 2021-02-09:
I see [bind.js:setupReporting()](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/bind.js#L277), called when the bulb is added to the group that triggers reporting to be set up, first binds the endpoint to the coordinator, and [bind.js:disableUnnecessaryReportings()](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/bind.js#L304) unbinds it when reporting is disabled.

But [bridge.js:deviceConfigureReporting()](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/bridge.js#L340), called when a `bridge/request/device/bind` MQTT message is received, does not bind the cluster. Is there a reason that it doesn't first bind to the endpoint (and unbind when removing reporting) like the other function?

Koenkk, 2021-02-10:
Added the bind, can you check if this is fixed now?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

crowbarz, 2021-02-11:
Reporting is now working on 1.17.1-dev @ 084d6ba - thanks!

As a side note, the code as is won't unbind from the cluster when `maximum_report_interval` is set to `65535` to remove the reporting. But you only want to do that if there are no other attributes still reporting on that cluster, which is what I think `bind.js:disableUnnecessaryReportings()` does. In theory you should be able to unbind the cluster manually via the API, though I haven't been able to get that to work. I will test binding later and log a separate bug for that once I can reproduce reliably.

closed 2021-02-11 by crowbarz