Koenkk/zigbee2mqtt #6194: LIDL bulp Only one effect working and Transition not work

shirou93, 2021-02-09:
## What happened
Propably one effect work in list. When try choise random effect bulp only blink

    effect_list: 
      - "blink"
      - "breathe"
      - "okay"
      - "channel_change"
      - "finish_effect"
      - "stop_effect"

Transision push in mqtt also don't work brightness is workig but when add {"brightness":156,"transition":3} change transision no react. Working the same.

Propably issue is on all lidl bulps.

## What did you expect to happen
Working effects.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:  1.17.1 
Adapter hardware: CC2652RB
Adapter firmware version: 
zStack3x0 20210120

Koenkk, 2021-02-09:
Besides hiding this from the docs probably nothing we can do from the Z2M side here because it looks like the firmware doesn't support it.

shirou93, 2021-02-09:
> Besides hiding this from the docs probably nothing we can do from the Z2M side here because it looks like the firmware doesn't support it.

`{"id":25,"type":"Router","ieeeAddr":"0x842e14fffe1b95c6","nwkAddr":39022,"manufId":4098,"manufName":"_TZ3000_dbou1ap4","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_dbou1ap4","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1612887497343}`

closed 2021-03-19 by github-actions[bot]