Koenkk/zigbee2mqtt #6196: LIDL bulp HG06106C Failed to setup reporting

shirou93, 2021-02-09:
## What happened
error on log
`Zigbee2MQTT:error 2021-02-09 15:28:43: Failed to setup reporting for '0x842e14fffe1b95c6' - Error: Read 0x842e14fffe1b95c6/1 lightingColorCtrl(["colorCapabilities"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:251:26)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)`
## What did you expect to happen

clear warn logs.
## How to reproduce it (minimal and precise)
pair device to z2m and restart z2m

## Debug info
Zigbee2MQTT version:  1.17.1 
Adapter hardware: CC2652RB
Adapter firmware version: 
zStack3x0 20210120

Koenkk, 2021-02-09:
Can you share the data/database.db entry of this device?

shirou93, 2021-02-09:
> Can you share the data/database.db entry of this device?

``{"id":25,"type":"Router","ieeeAddr":"0x842e14fffe1b95c6","nwkAddr":39022,"manufId":4098,"manufName":"_TZ3000_dbou1ap4","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_dbou1ap4","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1612887497343}``

Koenkk, 2021-02-09:
Can you change it to 

```
{"id":25,"type":"Router","ieeeAddr":"0x842e14fffe1b95c6","nwkAddr":39022,"manufId":4098,"manufName":"_TZ3000_dbou1ap4","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_dbou1ap4","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":15}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8eae9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1612887497343}
```

and try again?

shirou93, 2021-02-09:
> Can you change it to
>
> ```
> {"id":25,"type":"Router","ieeeAddr":"0x842e14fffe1b95c6","nwkAddr":39022,"manufId":4098,"manufName":"_TZ3000_dbou1ap4","powerSource":"Mains (single [...]

no error in logs after change this line.

Koenkk, 2021-02-09:
Should be permanently  fixed now (try to remove the device and repair to confirm)

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

shirou93, 2021-02-09:
Thanks. Propably all bulbs from Lidl needs this fix.

Koenkk, 2021-02-10:
Did that already. Assuming this can be closed now.

closed 2021-02-10 by Koenkk