Koenkk/zigbee2mqtt #6197: Interview failed for Osram Smart+ Plug

rickieyyy, 2021-02-09:
## What happened
When I try to connect the Osram smart+ plug, I get an error saying the interview has failed

## What did you expect to happen
The device is support, so it should be detected: https://www.zigbee2mqtt.io/devices/AB3257001NJ.html

## What I've tried (to fix)
- Restart Z2M.
- Put it close to the stick (within 10cm)
- Use touchlink
- Reset the plug by switching it on for 5sec and off for 5sec and repeat it 5 times so the plug resets.
- Removing it, and force removing it and reconnectiong by holding the plug 10 sec

## How to reproduce it (minimal and precise)
I hold the plug button for 10 sec, after that the plug pairs but the interview fails, due to a timeout.

## Debug info
I tried to print the error:
```
  zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x7cb03eaa0a09d3c1', attempt 6 +23s
  zigbee-herdsman:controller:device:log Interview - failed for device '0x7cb03eaa0a09d3c1' with error 'Error: Interview failed because can not get node descriptor ('0x7cb03eaa0a09d3c1')
  zigbee-herdsman:controller:device:log     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:350:23)
  zigbee-herdsman:controller:device:log     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device:log     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0x7cb03eaa0a09d3c1 with error 'Error: Interview failed because can not get node descriptor ('0x7cb03eaa0a09d3c1')' +29m
Zigbee2MQTT:error 2021-02-08 13:05:57: Failed to interview '0x7cb03eaa0a09d3c1', device has not successfully been paired
```
after 6 times:
```
  zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x7cb03eaa0a08f927', attempt 6 +1ms
  zigbee-herdsman:controller:device:log Interview - failed for device '0x7cb03eaa0a08f927' with error 'Error: Interview failed because can not get node descriptor ('0x7cb03eaa0a08f927')
  zigbee-herdsman:controller:device:log     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:351:23)
  zigbee-herdsman:controller:device:log     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device:log     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0x7cb03eaa0a08f927 with error 'Error: Interview failed because can not get node descriptor ('0x7cb03eaa0a08f927')' +3m
Zigbee2MQTT:error 2021-02-09 13:32:31: Failed to interview '0x7cb03eaa0a08f927', device has not successfully been paired
```

shirou93, 2021-02-09:
adapter?

rickieyyy, 2021-02-09:
@shirou93 The cc2531 with the latest firmware

closed 2021-03-19 by github-actions[bot]