Koenkk/zigbee2mqtt #6202: RTCGQ11LM low illuminance values

eherranzr, 2021-02-09:
## What happened
Illuminance and illuminance_lux values with RTCGQ11LM are very low. Most of the time they are even 0.

## What did you expect to happen
Values of illuminance and illuminance_lux on a reasonable scale taking into account that 0 lux is almost impossible to achieve

## How to reproduce it (minimal and precise)
Sensors are left still on a normal illuminated room where a normal person can see normally

## Debug info
Zigbee2MQTT version: 1.17.1 (via docker)
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

As explained above, illuminance and illuminance_lux values are always 0 or up to no higher than 8 on a normally lighted room. Values only increase when directly under a light. Mobile phone flash directedly towards the sensor shows around 700lx.

I tried to extract the cover of the sensor but values are still the same. Cover seems not to affect.

Logs can be found on [https://pastebin.com/u2fXBKha](https://pastebin.com/u2fXBKha)

If any further information or log is needed, please let me know and I'll share as soon as possible

Koenkk, 2021-02-10:
Looks similar to https://github.com/Koenkk/zigbee-herdsman-converters/pull/2166 @rikvb 

Following up on: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2166#issuecomment-768900012 I ran some tests
- The max illuminance value I get is always exactly 700, it's impossible to get a higher value
- I put it next to my Hue outdoor sensor (positioned outdoor) which measures +- 2019 lx. According to [Wikipedia](https://en.wikipedia.org/wiki/Lux) this would classify it a bit lighter than an overcast day (which seems right).  My RTCGQ11LM measures 700 lx (the max)

rikvb, 2021-02-10:
I've done some tests as well with several of these devices and the highest number I got was 776. This is with devices in window sills and in door openings. Previous values I reported in #2166 were all from inside the house.

Putting 776 into the formula gives around 29k, which doesn't sound too strange looking at that wikipedia article. Guess we'll have to wait for a sunny day, because that should definately give 5-digit values.

Koenkk, 2021-02-10:
@rikvb what values do you get when using e.g. the flaslight of your phone and holding it against the illuminance sensor? (I get 700 with mine).

rikvb, 2021-02-10:
Just tried it with one and got 636

eherranzr, 2021-02-10:
If it helps, I bought both affected sensors on November at AliExpress. I tried the flashlight on one of them and I reach 750 (looks to be the maximum as I tried twice).
What is weird, today at midday sensor seemed to work for a couple of hours until lx value went again back down to zero with no light change. Last value is the flashlight test.
[image]

arfond, 2021-02-12:
I wouldn't rely on this device to provide reliable illuminance measurements. It is measuring light to simply work out if it's day or night, allowing users to switch lights for example only at night when presence is detected. 700 lux max is more than enough to work out if it's day or night.

For those working in lighting (like me) a "typical overcast sky" is 10,000 lux. It varies massively with time of day and year though.

I have a GZCGQ01LM, which is more reliable. It has a range up to 83,000 lux. Not enough to reach summer sunny peaks of 100,000 lux or more, but it correlates well with the "professional" illuminance meter I have.

maddocche, 2021-03-30:
Don't know if related, but I have 3 of these sensors and one of them reports lux values extremely low compared to the other two. Like 30 lux inside a room in a sunny day; the other ones reports about 120 lux.
However pointing the phone flashlight at the "faulty" one gives a reading of 760 lux.

Is it an "hardware" faulty one or could it be something at software level? Is there a way to tell?

Koenkk, 2021-03-30:
@maddocche very strange, I expect that diff is at the software or hardware level of the sensor.

maddocche, 2021-03-30:
@Koenkk I think so, but I don't know how to test it. Do you know if there's a way to check/update the firmware or even an hardware disassembly guide? Some time ago I found the one to modify the hardware to increase the reporting frequency, but don't know if it can help.

Again, sorry if I use this thread improperly, if there's a better place for this discussion please tell me and I'll ask there! Thanks!

Koenkk, 2021-03-31:
@maddocche maybe it has something to do with the firmware (which cannot be updated).

Can everyone share their data/database.db entries of this device?

maddocche, 2021-03-31:
Ok, my "faulty" one is the first of the three.

[lumi_sensor_db.log](https://github.com/Koenkk/zigbee2mqtt/files/6238348/lumi_sensor_db.log)

LucidityCrash, 2021-04-21:
I've just got an RTCGQ11LM and a stand alone lux sensor  ( GZCGQ01LM ) 
and in the same location most of the time the RTCGQ11LM reads higher than the GZCGQ01LM  by a significant amount but it seems to ignore the illuminance_lux_calibration setting so I can't align them also when I shine a light on them the RTCGQ11LM seems to max out at 714 lx

debug 2021-04-21 11:40:35: Received Zigbee message from 'PIR 1', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":714}' from endpoint 1 with groupID 0
debug 2021-04-21 11:40:42: Received Zigbee message from 'Lux 1', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":33073}' from endpoint 1 with groupID 0

The  GZCGQ01LM  reports illuminance so that 33073 appears to convert to 2029 lux
 
database.db entries : 
The RTCGQ11LM 
{"id":18,"type":"EndDevice","ieeeAddr":"0x00158d0001dfc318","nwkAddr":48368,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,65535,1030,1024,1280,1,3],"outClusterList":[0,25],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":714}},"msOccupancySensing":{"attributes":{"occupancy":1}},"genBasic":{"attributes":{"65281":{"1":3045,"3":21,"4":12712,"5":9,"6":[0,7],"10":5040,"11":96,"100":0},"manufacturerName":"LUMI","zclVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20170627","swBuildId":"3000-0001","modelId":"lumi.sensor_motion.aq2"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":5,"stackVersion":2,"hwVersion":1,"dateCode":"20170627","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1619001101455}

The GZCGQ01LM  
{"id":14,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c7d114d","nwkAddr":7440,"manufId":4718,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sen_ill.mgl01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1024,3,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.sen_ill.mgl01","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20191118","swBuildId":"2019\u0000www."}},"genPowerCfg":{"attributes":{"batteryVoltage":31}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":27110}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b7790f","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b7790f","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1024,"attrId":0,"minRepIntval":15,"maxRepIntval":3600,"repChange":500}],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20191118","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1619001100016}

Hope that helps some :)

D43m0n, 2021-05-04:
I too see pretty low illuminance_lux values, I expected them to be higher. I've just paired a RTCGQ11LM today. I've also got a Philips Hue Motion Sensor, when I place them next to each other, they do both seem to report about the same current level however:

- RTCGQ11LM: `illuminance_lux: 66`
- Hue Motion Sensor: `54`

But both vary quite a bit, I'll have them sitting next to each other for a period of time. From the past 30 minutes I'd say the Hue motion sensor mostly has a bit lower reading than the Xiaomi/Aqara motion sensor.

I'd expected the RTCGQ11LM at least "see" when it's daylight, although it's cloudy outside and raining every once in a while. But when I place it in the hallway, (not the brightest place in our home) the illuminance_lux reading is often 0 (zero), well, it's still daytime, the sun is still up 😄

closed 2021-06-12 by github-actions[bot]

matejdro, 2021-11-09:
I actually have a reverse issue:

One of my sensors shows 0 when there is complete darkness outside, but then as soon as first ray of sunshine appears (when it is still dark enough that lights need to turn on), it shoots up to 14 and stays there for a better part of the day. 

it might be a faulty unit but I've just given up on these lux measurements and I'm using the sun elevation to roughly determine when to turn on lights.

XRedShark, 2024-11-01:
Mine has completely senseless values. During the day it shows 0 lux, at night it even reaches 30 lux, but the only thing that is on is the modem LED.