Koenkk/zigbee2mqtt #6211: New device - TS0601 - _TZE200_zion52ef

MarekCoufal, 2021-02-10:
Good day,
I bought a thermostat for Aliexpress. And I can't add it to the Zigbee network.
[https://www.aliexpress.com/item/1005001665618389.html?spm=a2g0s.9042311.0.0.7fda4c4dVskXSk](https://www.aliexpress.com/item/1005001665618389.html?spm=a2g0s.9042311.0.0.7fda4c4dVskXSk)

cat ./database.db

{"id":2,"type":"EndDevice","ieeeAddr":"0x5c0272fffec20dfb","nwkAddr":29060,"manufId":4098,"manufName":"_TZE200_zion52ef","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"�T�'\u0012�T�'\u0012�Y�'\u0011","65506":27,"modelId":"TS0601","manufacturerName":"_TZE200_zion52ef","powerSource":3,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1612951058691}

Thank you for your help
Marek

olili, 2021-02-20:
same problem. Can anybody help?

closed 2021-03-31 by github-actions[bot]

olili, 2021-10-10:
issue reactivated, in between any progress?