Koenkk/zigbee2mqtt #6212: Supported Devices Documentation: show if device data is pushed or pulled

SVNKoch, 2021-02-10:
Lately I've looked into getting a Zigbee Plug that also meassures power.
I've came to read the following issue: #3812 and saw that the Blitzwolf Plug does not push the state when it changes but it rather has to be pulled in an interval (like 10, 30 or 60 seconds). (In this issue one can find different configurations for different intervals.)

As far as I know there are however also devices that push their new state if it changes like the Xiaomi Aqara temperature, humidity and pressure sensor.
For some usecase (eg the kettle detection use case mentioned by [yourself](https://github.com/Koenkk/zigbee2mqtt/issues/3812#issuecomment-651655389)) the timing is critical and thus a pull might happen 9/29/59 seconds "too late" as the interval was just passed.

I propose updating the documentation so it shows if the attribute is updated via push or pull and if using pull, on what interval.

Thanks for you awesome work!

Koenkk, 2021-02-10:
The only 2 support power plugs that are being polled are https://www.zigbee2mqtt.io/devices/TS0121_plug.html and https://www.zigbee2mqtt.io/devices/HS2SK_nxp.html , for both it's documented already under the https://www.zigbee2mqtt.io/devices/TS0121_plug.html#device-type-specific-configuration

SVNKoch, 2021-02-25:
somehow I missed that the Blitzwolf are Tuya ones. And that these already have the information displayed there 😅
Thats great news! Thank you so much!

closed 2021-02-25 by SVNKoch