Koenkk/zigbee2mqtt #6213: Data request failed with error: 'No network route' (205))

darkkatarsis, 2021-02-10:
## What happened
After updating to the latest version, the network map does not show the connection between devices. LIQ for most devices is N / A and the console shows the error:

```
Zigbee2MQTT:error 2021-02-10 15:15:06: Publish 'set' 'state' to 'office_blind_roller' failed: 'Error: Command 0x680ae2fffe43ee48/1 closuresWindowCovering.upOpen({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
```

Downgrading to CC2652RB_20201026.hex (my previous version) also does not fix the problem.

Should I wait for the network to rebuild? Several devices are still working

## Debug info
Zigbee2MQTT version: 1.16.2.2
Adapter hardware: CC2652RB from Slash
Adapter firmware version: CC2652RB_coordinator_20210120.hex

Koenkk, 2021-02-10:
Repowering some devices usually fixes this.

darkkatarsis, 2021-02-10:
Unfortunately, it did not help. But the repairing did

closed 2021-02-10 by darkkatarsis