Koenkk/zigbee2mqtt #6222: Reconfigure reporting after firmware upgrade

sjorge, 2021-02-10:
## Describe the solution you'd like
after upgrade my FLOALT panels to the new ZB3 firmware reporting stopped working. Hitting apply again fixed it.

I don’t have any more bulbs with pending updates, but we should look into wether we need to re-setup reporting or not after and OTA update.

I believe we already re-setup reporting for trådfri devices on deviceAnnouncement so I wonder why I had to manually hit apply again. Maybe something in the update flow skips it?

## Describe alternatives you've considered
Stop being lazy and hit the apply button after every OTA update

## Additional context
Add any other context or screenshots about the feature request here.

sjorge, 2021-02-11:
I forgot to reapply for one panel and cutting the power and restoring works too... I wonder if the deviceAnnouncement was not send after the firmware for upgrade 🤔

MattWestb, 2021-02-11:
Was you doing the firmware updating without sniffing every package being sent to / from your panels ?!?!? 

More serious:

The original firmware was ZLL and the new is one ZB3 and its having new / changed endpoints and clusters. Perhaps one device announcement is enough for Z2M reading and doing the new configuration of the device.
But to being on the safe side i think its better rejoining the device then its for the system one completely new device (except IEEE and name).

How is you Zigbbee 3 panels working after the update ?

sjorge, 2021-02-11:
I wasn't sniffing traffic while updaten, aside from missing reporting they seem to be working fine... even without rejoining them.

I think z2m updated the clusters automatically:
[image]

The haDiagnostic seems new, at least I haven't seen it before. 

The bug where some would not turn on via a brightness change via scenes seems to be fixed too.

sjorge, 2021-02-11:
Looks like it also forgot it's scenes

MattWestb, 2021-02-11:
Most old IKEA ZLL devices was having diagnostic cluster but not new ones but its not written in stone.
My IKEA devices is only the old CWS that is haviing the diagnostic cluster all other dont have it.

I thinking IKEA have putting it back after the "stack blocking" bug that Silabs is having in the SDK until 6.7.8.0.

You is getting better router functions in the network with ZB3 routers and hopefully  better stability and lesser bugs.

One false thing `"zclVersion": 1,` shall being  2 for ZB3 devices :-)

MattWestb, 2021-02-11:
Did the OTA restoring the "default scenes in the devices ??

sjorge, 2021-02-11:
> Did the OTA restoring the "default scenes in the devices ??
> 
> Sorry you have already making new ones !!

They are not paired with an ikea remote, so not sure.

closed 2021-03-21 by github-actions[bot]