Koenkk/zigbee2mqtt #6228: Moes ZK-EU-2U plug USB not identified

FaBRiK74, 2021-02-11:
I just added a Moes ZK-EU-2U to my network. It has 2 exposes: Endpoint: l1 and Endpoint: l2
Unfortunately the Endpoint: l2 doesn't work via Z2M. It means when I try to switch it wia Z2M it gives an error and when I switch it on the device it gives "state_undefined" instead of "state_l2".
The logs:
MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe932122', payload '{"linkquality":123,"state_l1":"OFF","state_undefined":"ON"}'

Error Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'getDevice' of undefined at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.17.1/lib/extension/publish.js:174:43) at Controller.callExtensionMethod (/zigbee2mqtt-1.17.1/lib/controller.js:382:44) at runMicrotasks (<anonymous>) at processTicksAndRejections (internal/process/task_queues.js:93:5))

infeeeee, 2021-02-13:
I think you have the new version, I just also wanted to open an issue about this. 

How your plug identified?  Is it Model HGZB-01, Vendor: Nue / 3A?

If yes, than we  had a discussion about this here: https://github.com/Koenkk/zigbee2mqtt/discussions/5817 The symptoms are similar than yours. There are more detailed information and logs in the discussion.

I disassembled my device, this version information is written on the board:

```
XN-WGCZ+2USB T1.3
2020.06.04
```

Scroll down on the discussion to see photos of the interior. It seems like a new revision.

If your problem is different, I will open a new issue for my problem.

FaBRiK74, 2021-02-13:
It is identified as TS0112, Model ZK-EU-2U. Vendor is Moes.
I can switch on and off both (plug and USB) with the button on the device, but I can Switch on/off only the plug via Z2M. And it is pretty sure that the reason that l2 is identified wrong.

infeeeee, 2021-02-13:
Ok, than your issue is different, I will open a separate issue.

I will ping you and ask a question in the linked thread, but it's offtopic, so not here.

Koenkk, 2021-02-14:
@FaBRiK74 can you share the data/database.db entry of your device?

FaBRiK74, 2021-02-14:
@Koenkk Sure: {"id":4,"type":"Router","ieeeAddr":"0xec1bbdfffe932122","nwkAddr":48229,"manufId":4098,"manufName":"_TYZB01_3dbhmeky","powerSource":"Mains (single phase)","modelId":"TS0112","epList":[1,7],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0112","manufacturerName":"_TYZB01_3dbhmeky","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e297a2","endpointID":1}],"configuredReportings":[],"meta":{}},"7":{"profId":260,"epId":7,"devId":81,"inClusterList":[4,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613309372885}

Koenkk, 2021-02-14:
Can you try changing the devices.js definition to below and check if it works?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

```js
{
    zigbeeModel: ['TS0112'],
    model: 'ZK-EU-2U',
    vendor: 'Moes',
    description: 'Zigbee 3.0 dual USB wireless socket plug',
    extend: preset.switch(),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
    meta: {multiEndpoint: true, configureKey: 1},
    endpoint: (device) => {
        const hasEndpoint2 = !!device.getEndpoint(2);
        return {l1: 1, l2: hasEndpoint2 ? 2 : 7};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
},
```

FaBRiK74, 2021-02-14:
I know I'm a looser, but I couldn't find devices.js. Even the path. I have HA on RPi and I also have ssh. I can see the following diretories: addons, backup, config, media, share, ssl
No one of the contains devices.js

oleynikandrey, 2021-02-14:
@FaBRiK74, try `sudo find / -type f -name devices.js`

FaBRiK74, 2021-02-14:
@oleynikandrey I'm totally noob. Where shoul I try this command?

oleynikandrey, 2021-02-14:
> @oleynikandrey I'm totally noob. Where shoul I try this command?

`device.js` file you are looking for is located on RPi. So you need to ssh to it and execute the command.

FaBRiK74, 2021-02-15:
@oleynikandrey Thanks for the help. Unfortunately it gives "Connection refused"

Koenkk, 2021-02-15:
Since I have good confident it will fix your issue I've added the fix. Please let me know if it works

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

FaBRiK74, 2021-02-15:
@Kosta-Github Works perfectly. Thank you so much! :)

closed 2021-02-15 by FaBRiK74