Koenkk/zigbee2mqtt #6238: RTCGQ11LM double reporting any occupancy,true events

22nra, 2021-02-12:
## What happened
Have been using Z2M for sometime with Xiaomi sensors and all working well.
Recently added the PiR with lux RTCGQ11LM and noticed that although sensor is working it appears to be double reporting with the same occupancy,true event being repeated twice. Have the RTCGQ01LM (without lux) which is working fine.

## What did you expect to happen
Single report of occupancy,true event as seen with the RTCGQ01LM

## How to reproduce it (minimal and precise)
Have sensor RTCGQ11LM pointed at wall and move hand across to trigger one event but double event is reported.
See below double event with same time at end of log.txt at time 2021-02-12 08:03:06: 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5970975/log.txt)

## Debug info
Zigbee2MQTT version: 1.17.1 (commit #99274c7)
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20201127

configuration.yaml
homeassistant: false
permit_join: false
frontend:
  port: 8080
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  include_device_information: true
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d00024148df':
    friendly_name: PiR-Kitchen
  '0x00158d0003583474':
    friendly_name: Front-Door
  '0x00158d000464e1f7':
    friendly_name: TmpHum
  '0x00158d0005228f31':
    friendly_name: PiRlux-Liv-Rm
advanced:
  last_seen: disable
experimental:
  output: json

Koenkk, 2021-02-12:
This is expected since it fixes https://github.com/Koenkk/zigbee2mqtt/issues/2274 .

TLDR; sensor sends two messages on motion, illuminance and occupancy. Since some routers don't pass the occupancy message occupancy: true will also be send on the illuminance report. 

A debounce of e.g. 0.1 could solve this (see https://www.zigbee2mqtt.io/information/configuration.html#all-devices)

22nra, 2021-02-14:
Many thanks Koen, the debounce option makes sense and it has fixed the issue :)

It must be very difficult supporting so many devices that have different reporting methods. 
Just in case anyone else has this problem see below my devices configuration.yaml with the added debounce option.

devices:
  '0x00158d0005228f31':
    friendly_name: PiRlux-Liv-Rm
    debounce: 0.1

closed 2021-02-14 by 22nra