Koenkk/zigbee2mqtt #6239: Light is automatically turning on (sometimes) if automation has turned it off

ceesvanegmond, 2021-02-12:
Hi all,
I've a weird problem using zigbee2mqtt in Home Assistant where my light is turning itself on after it's turned off by a automation

- Home Assistant 2021.2.3
- Zigbee2mqtt latest (1.17.1-1)
- HA runs in VM using Slaesh's CC2652RB stick
- Motion detection Xiaomi human
- Kitchen lights are 3 grouped Ikea Tradfri GU10 lights (grouped in Zigbee2mq2tt to have 1 entity in HA)

```yaml
---
platform: mqtt
schema: json
name: kitchen
command_topic: "zigbee2mqtt/Kitchen/set"
state_topic: "zigbee2mqtt/Kitchen"
color_temp: true
brightness: true
```

So I have this simple automation for turning kitchen light on which is working flawlessly.

```yaml
---
alias: Lampen - Turn on kitchen lights by movement
description: ''

trigger:
  - entity_id: binary_sensor.kitchen_motion_occupancy
    platform: state
    from: 'off'
    to: 'on'

condition:
  condition: or
  conditions:
    - condition: numeric_state
      entity_id: sensor.kitchen_motion_illuminance_lux
      below: 101

action:
  - service: light.turn_on
    data:
      entity_id: light.kitchen
```

And this automation to turn light off when no motion detected:

```yaml
---
alias: Lampen - Turn off kitchen light
description: ''

trigger:
  - entity_id: binary_sensor.kitchen_motion_occupancy
    platform: state
    to: 'off'
    for:
      minutes: 1

condition: []

action:
  - service: light.turn_off
    data:
      entity_id: light.kitchen
```

Weirdly; sometimes when the light turns off automatically by the automation; the light is turning on again. And I don't know why. When checking out the logbook in HA something is weird. As the screenshot says: 'Keuken (which is kitchen in English' turned on by service.light.turn_on'. So that seems to me that there's no other automation that is triggering this randomly turned off light.

[image]

These are the zigbee2mqtt logs. The first line is automation that's beiing triggered to turn the light off. A few seconds after that; for some reason the light is turning on itself.

```log
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"state":"OFF"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 2', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":57,"state":"OFF","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 3', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":78,"state":"OFF","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Livingroom 1', payload '{"brightness":36,"color_temp":500,"linkquality":54,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 1', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":87,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 2', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":57,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 3', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":78,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 1', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":87,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 2', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":57,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 3', payload '{"brightness":145,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":78,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 1', payload '{"brightness":144,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":87,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"brightness":144,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 2', payload '{"brightness":144,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":57,"state":"ON","update":{"state":"idle"},"update_available":false}'
INFO: MQTT publish: topic 'zigbee2mqtt/Kitchen 3', payload '{"brightness":144,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":78,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Does someone has any clue? How can I debug this even further?

ceesvanegmond, 2021-02-16:
Nobody?

Koenkk, 2021-02-17:
The debug log should show you wether Zigbee2MQTT received an MQTT message (command) to turn off the light. If that is the case you can exclude Zigbee2MQTT from being the issue here.

ceesvanegmond, 2021-02-23:
@Koenkk Thanks for this first step. I added debug mode and this is the log.

[image]

What we can see in the first RED box is a received message with state OFF to the group Kitchen. In that Kitchen group are kitchen 1, 2 and 3. But only kitchen 1 and 2 are beiing shutdown; not kitchen 3.  Which is weird but because when kitchen light is turned on; all 3 are beiing turned on. Weird no?

Koenkk, 2021-02-24:
That's weird indeed. Is this also reproducible in the latest-dev? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) 

If yes provide the debug log when
- Starting zigbee2mqtt
- Immediately after this reproduce the bug

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

ceesvanegmond, 2021-02-26:
@Koenkk Seems it's not part of zigbee2mqtt but has to do with 'Adaptive lightning'. https://github.com/basnijholt/adaptive-lighting/issues/85

closed 2021-02-26 by ceesvanegmond

CodeFinder2, 2025-10-25:
> [@Koenkk](https://github.com/Koenkk) Seems it's not part of zigbee2mqtt but has to do with 'Adaptive lightning'. [basnijholt/adaptive-lighting#85](https://github.com/basnijholt/adaptive-lighting/issues/85)

Thanks for the pointer! I have the same problem and it may be related to the official [Flux](https://www.home-assistant.io/integrations/flux/) integration. So quite similar.