Koenkk/zigbee2mqtt #6242: Linkind Keypad ZS130000078

HASSHUN, 2021-02-12:
I'd like to ask you to help me integrating this keypad to the database. I planned to calibrate it to my alarm_control_panel, so make it able to follow commands like arm_away, arm_home and disarm.

First let's start with the configuration, cause that also can contain mistakes:
For examples, I used Xfinity and 3400-D keypads' inputs cause they looked similar (except for the fact that they are standalone devices). Here is my conf in device.js [1.17.1 (99274c7)]
```
    {
        zigbeeModel: ['ZB-KeypadGeneric-D0002'],
        model: 'ZS130000078',
        vendor: 'Linkind',
        description: 'Security keypad battery',
	meta: {configureKey: 1, battery: {voltageToPercentage: ['3V_2100'}},]
	fromZigbee: [fz.command_arm, fz.battery, fz.ias_sos_alarm_2], 
        exposes: [e.battery(), e.voltage(), e.battery_low(), 
            exposes.numeric('action_code', ea.STATE), e.action(['disarm', 'arm_all_zones', 'invalid_code', 'emergency'])],
    	toZigbee: [tz.arm_mode], 
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            const clusters = ['genPowerCfg', 'ssIasZone', 'ssIasAce', 'genBasic'];
            await reporting.bind(endpoint, coordinatorEndpoint, clusters);
            await reporting.batteryVoltage(endpoint);
        },
        onEvent: async (type, data, device) => {
            if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' &&
                globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
                const payload = {
                    panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                    secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
                };
                await device.getEndpoint(1).commandResponse(
                    'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
                );
            }
        },
    },

```
I managed to get it interviewed succesfully, but the keypad only communicates to my Conbee II device if it's not connected to the Siren hub. Once it is connected, the traffic just disappears ( probably switches channel or endpoint? ).
Payloads like arm or disarm also can only be commited when it's connected. 

All I can sniff is what happens between resetting the keypad and connecting to the hub. (see attached, filtered capture from Wireshark)

This issue seems very similar to Goodwin's case in #6174 with the Iris keypad. I'm trying to switch channels to be able to catch some communication between the keypad and the hub. 

Any ideas or hints are welcome!

## Information about the device + link
https://zigbeealliance.org/zigbee_products/linkind-keypad/
 The device has a hub and it is only available in a kit. 
https://www.linkind.com/product/smart-home-security-system-kit-ring-alarm/

## data/database.db entry of the device
..

HASSHUN, 2021-02-12:
Seems like the zip file is not uploaded properly. So here it is again.

Oh and about zigbee herdsmann debug here is capture as well.

Received Zigbee message from '0x680ae2fffee56c49', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID null
Received Zigbee message from '0x680ae2fffee56c49', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":822935558,"imageType":2308,"manufacturerCode":4456}' from endpoint 1 with groupID null

and it doesn't find a converter for Panelstatus:

Received Zigbee message from '0x680ae2fffee56c49', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID null
No converter available for 'ZS130000078' with cluster 'ssIasAce' and type 'commandGetPanelStatus' and data '{}'

[linkind_keypad_0x0B(11).zip](https://github.com/Koenkk/zigbee2mqtt/files/5971710/linkind_keypad_0x0B.11.zip)

kloodhu, 2021-02-13:
I try to configure the same device, and went a little but further.

A captured the communication between the original hub and keypad.
When i turn on the keypad (with the proximity sensor) it send a message to the hub:

[image]

The command "7" is the "getPanelStatus".

Then the hub after an ack message sent back to the keypad a response:

[image]

The command "5" is the "getPanelStatusRsp" with the payload "00 00 00 00".

I believe the keypad is waits for this response, but i don't know, how to write this into the tozigbee part.
The relevant code from the (original) toZigbee:

`    arm_mode: {
        key: ['arm_mode'],
        convertSet: async (entity, key, value, meta) => {
            const mode = utils.getKey(constants.armMode, value.mode, undefined, Number);
            if (mode === undefined) {
                throw new Error(`Unsupported mode: '${value.mode}', should be one of: ${Object.values(constants.armMode)}`);
            }

            if (value.hasOwnProperty('transaction')) {
                entity.commandResponse('ssIasAce', 'armRsp', {armnotification: mode}, {}, value.transaction);
            }

            let panelStatus = mode;
            if (meta.mapped.model === '3400-D') {
                panelStatus = mode !== 0 && mode !== 4 ? 0x80: 0x00;
            }
            globalStore.putValue(entity, 'panelStatus', panelStatus);
            const payload = {panelstatus: panelStatus, secondsremain: 0, audiblenotif: 0, alarmstatus: 0};
            entity.commandResponse('ssIasAce', 'panelStatusChanged', payload);
        },
    },`

A put this code to the js:

`const ias_panel_status = {
        cluster: 'ssIasAce',
        type: 'commandGetPanelStatus',
        convert: (model, msg, publish, options, meta) => {

            const panelStatus = msg.data.panelstatus;
            return {
                panelStatus
            };
        },
};`

I don't know this is necessary or just hide the error about no converter.

I can send the full capture with the key in private, if you think it is help.

HASSHUN, 2021-02-13:
Hi kloodhu,

That was a scenario ( panel wake up by proximity) I didn't look into, but mine only produces this after interviewing. Anyways, it still does not explain what happens after the hub is synced with the keypad. I'm unable to register a single line after it. Can you reach out anything related to arm or disarm?

@Koenkk or @goodwin or anyone from the mighty Gods of Z2M, based on the info we provided, can you give us a hint?

Koenkk, 2021-02-14:
The `getPanelStatus` can be done in the `onEvent`. The `onEvent` in the OP looks correct to me.

HASSHUN, 2021-02-15:
The panelstatus is the same at my side as well, but I think it's not the main issue that we don't manage to get the panelstatus converted.
Seems like the configuration for Xfinity or 3400-D does not match this device's. And the main problem is that I can't catch a single command from or to the keypad after it's synced with the hub.(the panelstatus thing kloodhu mentioned is also "off-hub" communication)

Do you know how to find out where does the communication flow now?

HASSHUN, 2021-02-15:
I see the device only under homeassistant in MQTT explorer. I tried to publish an action, but there is no identified topic for it. :I
[image]

HASSHUN, 2021-02-17:
@kloodhu is there any progress at your side?

HASSHUN, 2021-02-26:
@Koenkk Can you tell me please what does this mean below? I got this when the hub and the keypad are connected and I replace the batteries.

[image]

HASSHUN, 2021-03-15:
Any chance that someone would answer my question? I mean it is a great product and easily accessible across Europe. I'm sure if we could integrate this keypad many would benefit from it.

GSzabados, 2021-05-20:
Hello,

I would like to resurrect this topic. And make some further progress with some help from someone, who knows what has to be done.

I started from the OP's code, which had a typo, and I managed to reach some success, but I am confused and stuck.

So this is the OP's code with some minor changes, if you use it as an external converter. But unfortunately it is not working due to the onEvent's panelStatus part.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['ZB-KeypadGeneric-D0002'],
        model: 'ZS130000078',
        vendor: 'Linkind',
        description: 'Security keypad battery',
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        fromZigbee: [fz.command_arm, fz.battery, fz.ias_sos_alarm_2, fz.ias_ace_occupancy_with_timeout], 
        exposes: [e.battery(), e.voltage(), e.battery_low(), 
            exposes.numeric('action_code', ea.STATE), e.action(['disarm', 'arm_all_zones', 'invalid_code', 'emergency'])],
        toZigbee: [tz.arm_mode], 
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            const clusters = ['genPowerCfg', 'ssIasZone', 'ssIasAce', 'genBasic'];
            await reporting.bind(endpoint, coordinatorEndpoint, clusters);
            await reporting.batteryVoltage(endpoint);
        },
        onEvent: async (type, data, device) => {
            if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' &&
                globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
                const payload = {
                    panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                    secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
                };
                await device.getEndpoint(1).commandResponse(
                    'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
                );
            }
        },
     },
];
```

But if I remove some of it and hardcode the panelStatus, the panel wakes up and accept codes.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['ZB-KeypadGeneric-D0002'],
        model: 'ZS130000078',
        vendor: 'Linkind',
        description: 'Security keypad battery',
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        fromZigbee: [fz.command_arm, fz.battery, fz.ias_sos_alarm_2, fz.ias_ace_occupancy_with_timeout], 
        exposes: [e.battery(), e.voltage(), e.battery_low(), 
            exposes.numeric('action_code', ea.STATE), e.action(['disarm', 'arm_all_zones', 'invalid_code', 'emergency'])],
        toZigbee: [tz.arm_mode], 
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            const clusters = ['genPowerCfg', 'ssIasZone', 'ssIasAce', 'genBasic'];
            await reporting.bind(endpoint, coordinatorEndpoint, clusters);
            await reporting.batteryVoltage(endpoint);
        },
        onEvent: async (type, data, device) => {
            if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce') {
                const payload = {
                    panelstatus: 0x00,
                    secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
                };
                await device.getEndpoint(1).commandResponse(
                    'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
                );
            }
        },
     },
];
```

Indeed the panel starts to accept codes, when {     "arm_mode": {     "mode": "disarm"    } } like messages are passed to it with the non-hardcoded values as well, but I am not fully convinced what is the logic behind this panel, or how it really should work.
I am missing from the states of the device the panelStatus/armMode.

If someone has a better understanding how the panel should work, and what are the messages which should be passed from the coordinator to the panel when wakes up, that would make development a bit easier.

So far looks like, once awakes, it needs a panelStatus/armMode, in response to commandGetPanelStatus, but I don't know what is stored under panelStatus, as I cannot see it anywhere.

@Koenkk, could you re-open this issue please, and if you have any input, that would be appreciated.

GSzabados, 2021-05-20:
It really looks like, when paired a panelStatus 0x00 has to be sent first.

kloodhu, 2021-05-20:
The device is working almost perfect with ZHA now.

GSzabados, 2021-05-20:
> The device is working almost perfect with ZHA now.

Can you share some details what do you mean "almost"?

I will have a look how the code works there. 

It would be nice to make it work with Z2M as well. Just need to understand the quirks.

HASSHUN, 2021-05-24:
Hi @GSzabados  / Szia GSzabados!

In fact I just tried ZHA and everything works perfectly (no "almost" for me ), although I have to use a separate Zigbee receiver (CC2531) for that purpose, cause Z2M and ZHA cannot work in parallel on the same device. 

Nevertheless If you can find the right way to get your configuration pulled in, I'd switch back to Z2M with the keypads. So this issue is still not solved @Koenkk and as I see more users tends to be interested about the product. 

GSzabados,  If I can help with anything please let me know, I have all the means for it now,.(but lack the knowledge :P)

GSzabados, 2021-05-24:
@HASSHUN, I have a spare CC2531 as well, but I do not want to build another Zigbee network. I have already three of them, Z2M, Hue and SmartThings, and yesterday I redone my Z2M network with a CC2652P dongle after struggling for weeks with another firmware on the dongle. I reflashed to Koenkk's last firmware and re-paired everything. So no intention at the moment to have another for just the keypad.
But by looking at the ZHA solution, the keypad is recognised as an Alarm Panel as I can see from other posts, correct me if I am wrong. And the requirement for that is to hold a panel status or alarm status, I believe.
And I think that is missing from the Z2M setup.
As I can see the panel requires that and initial disarmed status is passed to it and then it needs to sync with the coordinator later on.
I cannot see in the current code that how is the initial disarmed is passed to the keypad. I cannot see from where it gets the panelStatus value for the onEvent response, unless a disarm command has passed first to the keypad, but I have not succeeded with that neither.

I saw an ongoing PR and issue for the Xfinity keypad which has had the panel status handled in some different way, but that PR hasn't been fully merged. https://github.com/Koenkk/zigbee-herdsman-converters/pull/1873
I have a guess, that method would be the right one. And Z2M should persist a state for the panel status, and not just the action what the button press provides.

I really need some help to understand what messages are passed to the device and what it is sending to the coordinator. And how to output that to a log. I turned on herdsman logging and debug for Z2M, but that didn't provide enough information. I would love to get some information how I can add some debug to the external converter. @Koenkk can you advise on that please?

HASSHUN, 2021-05-24:
@GSzabados.  I have a sniffer set , I'll look it up tomorrow morning what's happening through ZHA.

GSzabados, 2021-05-25:
> I really need some help to understand what messages are passed to the device and what it is sending to the coordinator. And how to output that to a log. I turned on herdsman logging and debug for [...]

@Koenkk, Never mind, I just realized, that the zigbee-herdsman debug is working, but it outputs only to the log (file), but not to the web interface's Log view. I think, I have now what I need to look at. :)

@HASSHUN, if you can provide some info how the pairing happens and what information ZHA passes to the keypad, when it wakes up, that would be the best.

GSzabados, 2021-05-25:
> @Koenkk, Never mind, I just realized, that the zigbee-herdsman debug is working, but it outputs only to the log (file), but not to the web interface's Log view. I think, I have now what I need to look at. :)

Actually, I have to correct myself. I see it in the Supervisor 's Add-ons' Log Tab, but I cannot see it in the log of Zigbee2MQTT in /share/zigbee2mqtt/log. Is it logged to file? Or just STDOUT? Or is there any way to save it to file?

Koenkk, 2021-05-25:
@GSzabados only logged to STDOUT.

GSzabados, 2021-05-25:
> @GSzabados only logged to STDOUT.

Thanks, then I will try some different way. I have already installed Z2M on a Windows machine a few days ago, just to be able to erase the NVMEM of a CC2652. I will try to do the work with that.

GSzabados, 2021-05-25:
> @GSzabados only logged to STDOUT.

How is this command should be used on a Windows installation?

`DEBUG=zigbee-herdsman* npm start`

I got only this:

```
'DEBUG' is not recognized as an internal or external command,
operable program or batch file.
```

GSzabados, 2021-05-25:
> > @GSzabados only logged to STDOUT.
> 
> How is this command should be used on a Windows installation?
> 
> `DEBUG=zigbee-herdsman* npm start`
> 
> I got only this:
> 
> ```
> 'DEBUG' is not recognized as an internal or external command,
> operable program or batch file.
> ```

I should have Google a bit first. If anyone comes along here to find it, the command is this for a Windows environment:

`setx DEBUG zigbee-herdsman*`

HASSHUN, 2021-05-26:
Hi @GSzabados ,

Seems like I've been a bit early stating everything is perfect with ZHA and the keypad.

It might be a general ZHA-raspbian issue, but by yesterday the CC2531 dongle had lost its signal and I was unable to repair it ever since then, no matter how I tried ( disabling / removing modem manager, reconfiguring ZHA integration ) . So no luck yet. I'll let you know when I manage to record something.

@Koenkk I know it's off-topic here, but I wonder does the CC2531 has newer firmware than the 20201127 one?

GSzabados, 2021-05-26:
@HASSHUN, no worries.

I've made some progress meanwhile I tried to do things through the Windows installation with the CC2531.

It has turned out that the globalStores constant were not included in the beginning of the external converter.

So for now this started to work:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const globalStore = require('zigbee-herdsman-converters/lib/store')
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['ZB-KeypadGeneric-D0002'],
        model: 'ZS130000078',
        vendor: 'Linkind',
        description: 'Security keypad battery',
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        fromZigbee: [fz.command_arm, fz.battery, fz.ias_sos_alarm_2, fz.ias_ace_occupancy_with_timeout], 
        exposes: [e.battery(), e.voltage(), e.battery_low(), 
            exposes.numeric('action_code', ea.STATE), e.action(['disarm', 'arm_all_zones', 'invalid_code', 'emergency'])],
        toZigbee: [tz.arm_mode], 
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            const clusters = ['genPowerCfg', 'ssIasZone', 'ssIasAce', 'genBasic', 'genIdentify'];
            await reporting.bind(endpoint, coordinatorEndpoint, clusters);
            await reporting.batteryVoltage(endpoint);
        },
        onEvent: async (type, data, device) => {
            if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' &&
                globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
                const payload = {
                    panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                    secondsremain: 0x00, audiblenotif: 0x00, alarmstatus: 0x00,
                };
                await device.getEndpoint(1).commandResponse(
                    'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
                );
            }
        },
     },
];

``` 

BUT!

You need to do publish the panel status first to:

zigbee2mqtt/[DEVICE_FRIENDLY_NAME/set

{
    "arm_mode": {
    "mode": "arm_all_zones" // Mode "arm_all_zones" or "disarm"
    }
}

GSzabados, 2021-05-29:
@Koenkk, I have made some progress with the keypad and functions seems to work without any issue.
The only problem what I see is with the tamper.

The device reports zoneStatus 52 when tampered and goes back to zoneStatus 48 when it is not tampered anymore. It uses supervision_report and restore_report as well. The only option which I found in fromZigbee is ias_smoke_alarm_1 with thise reports. Obviously not a right option with smoke alarm. 
Should I make a new function like ias_keypad_tamper? I have no clue what else the keypad can report. I guess it does report low_battery as well, but no idea about any alarms or test or trouble. Any advice on this matter?

The other things is the occupancy_timeout. How can I set it to 30 seconds in the converter? I haven't found any example for that.

Thanks for any input!

GSzabados, 2021-06-03:
> @Koenkk, I have made some progress with the keypad and functions seems to work without any issue.
> The only problem what I see is with the tamper.
>
> The device reports zoneStatus 52 when tampered and [...]

@Koenkk or @sjorge can you give me a little help with this? How it should be done?

GSzabados, 2021-06-12:
Bumping this to do not get closed. Unfortunately I haven't got time now to further test this, but I will try to make a PR. Basic functionality was working on my side but I will not have access to my test setup to do more.

thecobra666, 2021-06-15:
I have the xfinity version. Has there been any progress made on this? The integration in zha is very nice and stable. I to would love it if some more development could be done on this.

GSzabados, 2021-06-15:
> I have the xfinity version. Has there been any progress made on this? The integration in zha is very nice and stable. I to would love it if some more development could be done on this.

The Xfinity keypad is already supported.

https://www.zigbee2mqtt.io/devices/URC4450BC0-X-R.html

To reach full functionality you need to build Automations to respond to keypad event.

A few topics to have some idea what Automations you need to build:

https://community.home-assistant.io/t/how-to-get-the-secret-value-into-template/270600

https://community.home-assistant.io/t/how-to-use-the-mqtt-payload-in-action/269711

https://community.home-assistant.io/t/check-incoming-mqtt-message-for-correct-pin-then-respond-accordingly/260319

This is a deConz Blueprint, but gives some ideas:

https://community.home-assistant.io/t/deconz-simple-alarm-control-panel-blueprint/305397

Otherwise it might be a good idea to use Alarmo and the Alarm Panel status from there to control. (Just my though... I haven't tried it yet.) But just to note, Z2M adds the device to HA, but not as an Alarm Panel.

thecobra666, 2021-06-15:
> The Xfinity keypad is already supported.
>
> https://www.zigbee2mqtt.io/devices/URC4450BC0-X-R.html
>
> To reach full functionality you need to build Automations to respond to keypad event.
>
> A few topics [...]

I already use the automations but ZHA it was indeed an alarm panel which made automations a bit easier. I wonder if the MQTT manual alarm could be used to do the same: https://www.home-assistant.io/integrations/manual_mqtt/

GSzabados, 2021-07-22:
Just to keep it still open, I am still working on this.

stefangries, 2021-09-05:
For me, 1.21.0 broke the compatibility with the keypad. Issue seems related to [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/8630).

When trying to set the exit_delay mode, I get the following error in the zigbee2mqtt logs:
`Error Publish 'set' 'arm_mode' to 'Keypad' failed: 'Error: Unsupported mode: 'exit_delay', should be one of: disarm,arm_day_zones,arm_night_zones,arm_all_zones,invalid_code,not_ready,already_disarmed'`

GSzabados, 2021-09-05:
> invalid_code

Try invalid code... I need to check what exactly has changed.

GSzabados, 2021-09-05:
> When trying to set the exit_delay mode, I get the following error in the zigbee2mqtt logs

There was a PR which has changed a few things for keypads.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/2967

robertalexa, 2021-09-05:
> For me, 1.21.0 broke the compatibility with the keypad. Issue seems related to [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/8630).
>
> When trying to set the exit_delay mode, I get the [...]

Exactly that. Order of commands is different. Please read the issue you have quoted and also your keypad documentation

robertalexa, 2021-09-05:
@stefangries Follow my advice above, if you still have issues let's talk about it here https://github.com/Koenkk/zigbee2mqtt/issues/8630 (to keep things together and clean. But do not ask for help if you have not yet read that thread and the documentation. Most likely the answer to your problem is already written down)

@HASSHUN As the devices is integrated, can you please close this issue? Thanks :)

evilpie, 2021-09-05:
Is this device supported under a different model? I can't find anything when searching for `ZB-KeypadGeneric-D0002`.

robertalexa, 2021-09-05:
Also @HASSHUN and @GSzabados, I am very confused. Is the device in question this one? 
[image]

If the answer is yes, then please have a look at the Hive Keypad, as that is basically produced by Linkind, and it looks identical except for motion sensors at the front.

So try to copy the code from Hive into your custom converter and then adapt from there.

With the right order of commands you should be able to get this working really quick :)

GSzabados, 2021-09-05:
> LE: I am sorry, i got this thread confused with another one. My apologies, didn't mean to sound rude, just got confused.

Please excuse me, but yes, you were rude. I am rarely picking up this kind of comments, but your attitude was a bit off...

I could tell you the same, do not reply before reading the thread, but you already know that.

The integration is not finished. I am trying to work on it time to time, when I have time for it. Now I will have a look at your changes in the other PRs and try with that the external component.

Something is still missing regarding the IAS messages, and the device doesn't behave well time to time, but might be your change will fix that.

GSzabados, 2021-09-05:
> as that is basically produced by Linkind

Yes, that is the keypad. The producer is NOT Linkind. That is just an umbrella company like Frient for Develco. The developer and producer is Leedarson.

I will have a look on the Hive one and let you know.

robertalexa, 2021-09-05:
> Please excuse me, but yes, you were rude. I am rarely picking up this kind of comments, but your attitude was a bit off...

You are perfectly right and I can only apologise. I am currently looking after 4 threads regarding Alarm keypads, and I got this one confused for a completely different one, which was discussed in detail. I thought someone new came asking for the same question (e.g. being lazy with their research)

> Yes, that is the keypad. The producer is NOT Linkind. That is just an umbrella company like Frient for Develco. The developer and producer is Leedarson.

And you are correct again :) I even put this link in my PR https://www.leedarson.com/products/keypad
Indeed called Leedarson. Nonetheless, using Hive as a starting point is the best approach.

Feel free to ask for help if you get stuck

GSzabados, 2021-09-05:
> Feel free to ask for help if you get stuck

What is the _contact_ for on the Hive alarm panel?

robertalexa, 2021-09-05:
Absolutely no freaking idea. I wasn't able to properly test it using the hive ecosystem because i do not own any of their sensors. And you can do basically nothing from the hive app without sensors.

The contact does however report correcr values, but I wouldn't be able to tell you the logic behind it. I would suggest you use it for your Linkind keypad. If you get correct values, leave it in, even if we don't fully understand how they are generated

stefangries, 2021-09-06:
I now followed [the instructions for another keypad](https://www.zigbee2mqtt.io/devices/KEYPAD001.html#armingdisarming-from-the-keypad) and they helped to solve my problem. In short:

Never reply with action="exit_delay" to a message, but always with the same action and action_code as received. The keypad will never send action="exit_delay", so you shouldn't reply with it. Only after you correctly replied to the keypad message, you can send another message with action="exit_delay", but without an action_code.

robertalexa, 2021-09-06:
> but without an action_code.

Without a `transaction_id`. Apart from that you are correct, that is the intended flow :)

evilpie, 2021-09-06:
I spend some time coming up with a way for integrating the keypad with the "manual" [alarm control panel](https://www.home-assistant.io/integrations/manual) of home assistant. So far this seems to be working quite well: https://gist.github.com/evilpie/ce3d7b90b0584202b5efc4fe39d1d663. I am using the a state trigger for e.g alarm_control_panel `disarmed` to send `{ "arm_mode": { "mode": "disarm" } }` to the keypad. I would love for someone to review this? :smile: 

I am also still using the same device code in https://github.com/Koenkk/zigbee2mqtt/issues/6242#issuecomment-906565505. Maybe it's ready for a pull request?

thecobra666, 2021-09-08:
> I am also still using the same device code in #6242 (comment)

I am also doing the same. Have a lot of automations for it to work but, it
works :) based on the input I change the state of the keypad and the manual
alarm. Works great.

stefangries, 2021-09-15:
I would agree, that this is ready for a pull request. @GSzabados Is there anything still to fix or do you want to do a pull request?

robertalexa, 2021-09-15:
@stefangries I am willing to put a PR in for this device if @GSzabados is still busy. I am guessing you are using the code provided here: https://github.com/Koenkk/zigbee2mqtt/issues/6242#issuecomment-906565505

If so, could you try something please? Could you replace the `exposes` section with this:

```
exposes: [e.battery(), e.battery_voltage(), e.battery_low(), e.occupancy(), e.tamper(), e.contact(),
            exposes.numeric('action_code', ea.STATE).withDescription('Pin code introduced.'),
            exposes.numeric('action_transaction', ea.STATE).withDescription('Last action transaction number.'),
            exposes.text('action_zone', ea.STATE).withDescription('Alarm zone. Default value 23'),
            e.action([
                'panic', 'disarm', 'arm_day_zones', 'arm_all_zones', 'exit_delay', 'entry_delay'])],
```

That should expose a few more bits. If all is well, could you please confirm what is the value that you get for `action_zone` to make sure the attribute description is accurate.

Thanks

GSzabados, 2021-09-15:
@robertalexa, please, try to explain me, what is this for:

`e.contact()`

What Zigbee cluster is it using for this? And what triggers it? 

I am asking it, because it doesn't make any sense, unless there is a contact sensor (open/close) built into the device and that triggers on maybe opening the battery door or something.

robertalexa, 2021-09-15:
> Absolutely no freaking idea. I wasn't able to properly test it using the hive ecosystem because i do not own any of their sensors. And you can do basically nothing from the hive app without [...]

@GSzabados Already answered this above.

fz.ias_contact_alarm_1
[image]

This is something i saw in the Centralite alarm keypad and i have included in the Hive. As mentioned, the attribute gets exposed and populated with a value, but I do not understand either what is the logic behind it. My guess is that it goes hand in hand with the tamper sensor. Contact closed = tamper clear. I think tamper might be a "human friendly" indicator for the contact button that triggers the tamper.

Hope this helps

stefangries, 2021-09-15:
> ```
> exposes: [e.battery(), e.battery_voltage(), e.battery_low(), e.occupancy(), e.tamper(), e.contact(),
>             exposes.numeric('action_code', ea.STATE).withDescription('Pin code [...]

It works just as well as before - I couldn't tell any difference right away. The contact sensor also appears in Home Assistant - but I could not find a way to make the sensor switch. The tamper sensor works without any problems.

robertalexa, 2021-09-15:
@stefangries Sorry I was meant to mention checking the device in z2m Frontend. That would look similar to this:
[image]

Could you confirm the `action_zone` value exposed when using the device.

As to the contact entity, it is not a switch, as you can't SET the value, you can just GET it. I will try to dig out some information about it hopefully from the ZCL specs and report back.

GSzabados, 2021-09-15:
> I think tamper might be a "human friendly" indicator for the contact button that triggers the tamper.

Please check how the commandStatusChangeNotification in Zigbee messages.

Exposing `contact` is a wrong practice if the device doesn't report any. (It is always 0 or 1 in the set of IAS Notification messages.)

Look at this post here first.

https://github.com/Koenkk/zigbee2mqtt/issues/6242#issuecomment-850790129

stefangries, 2021-09-15:
> Could you confirm the `action_zone` value exposed when using the device.

The action_zone switches from None/Null in HomeAssistant/Zigbee2MQTT for an extremely short time when using the device to 23.

robertalexa, 2021-09-15:
> > Could you confirm the `action_zone` value exposed when using the device.
> 
> The action_zone switches from None/Null in HomeAssistant/Zigbee2MQTT for an extremely short time when using the device to 23.

Thanks for confirming. The same happens for the Hive one (same manufacturer). So the same `expose` section could be used.

@GSzabados I am not entirely sure what you are asking me at this point :) Have a look at the ZCL standard document in section 8. 2. 2. 1. 1. 2  ZoneType Attribute and 8. 2. 2. 1. 1. 3 ZoneStatusAttribute. You keep mixing tampering and contact thus confusing me a lot.
Tampering is reported in ZoneStatus
[image]

Whereas contact isn't a `status`, it's a `type`. But as mentioned before, I do not have a keypad anymore, so I can't test anything.  And I am unsure what would make the keypad report different `type` values. The decision is yours, use the `contact` attribute like all the other keypads, or don't :) There is no harm in doing either. It is also not bad practice since the device is actually reporting it. How it judges that is unknown by me, but the information is reported nonetheless.

I will leave this with you now, feel free to put in the PR when you are ready

GSzabados, 2021-09-15:
> fz.ias_contact_alarm_1

Is this what you posted above? Is it where the contact is extracted? Now have a look where is the tamper extracted from...

robertalexa, 2021-09-15:
Are you always passive aggressive? Cause I feel you don't understand sth, you ask for an opinion and then you act sassy :)

The answer is simple, use contact or don't, it does't make a difference. Just submit the PR and move on.

That will help all these people that are waiting on you :) After that, feel free to re-engineer how z2m handles the IASZone.

I tried to help, but it seems you got this, so I will leave you to it.

GSzabados, 2021-09-15:
> That will help all these people that are waiting on you :) After that, feel free to re-engineer how z2m handles the IASZone.

You don't understand how the device reports things and after you exposes entities (contact), what is not really reported by the device. 

Please read this post:

https://github.com/Koenkk/zigbee2mqtt/issues/6242#issuecomment-850790129

Please calculate what values the keypad reports in the zoneStatus and please do not mix tamper and contact.

Exposing contact without a real meaning, is like exposing luminance on a motion sensor which hasn't got a luminance sensor.

robertalexa, 2021-09-15:
You are a stubborn one aren't you.

I will says this for the 4th time now. Simply don't expose it and move on. I don't understand why you are still flapping about it. Remove the expose and do a PR :)

PS: you insist with that post about tampering yet you are talking about contact. Then you ask me not to mix the 2 together....

Just do what you want

[64 more replies]

closed 2021-10-04 by Koenkk