Koenkk/zigbee2mqtt #6243: Debouncing/filtering group attributes?

crowbarz, 2021-02-12:
## What happened
When device that is a member of a group is turned on, z2m seems to also update the group attributes with the device's current attributes, and sends an MQTT update if any of the group attributes have changed. This normally works ok except where some of the group member's attributes are not consistent across the group members (eg. colour, colour temp or level). In that case, multiple MQTT updates are sent for the group as the group attributes oscillate as each member's state transition to on is processed.

## What did you expect to happen
For the same reasons it is useful to [debounce/filter device updates]((https://www.zigbee2mqtt.io/information/configuration.html#device-specific-configuration)), it would be useful to be able to debounce and/or filter the attribute updates for groups.

## How to reproduce it (minimal and precise)
Say you have 9 bulbs in a group, with brightness all set to the same value. When the group is turned on, a single update is sent for the group with `"state": "ON"` and the current brightness value.

Now, set each of the bulb's brightness to a different value. Every time the group is now turned on, 9 updates are sent for the group, each with a different `"brightness"` value and in an apparently random order.

Only a single update for the group is sent when the group is turned off, even if the attributes differ. This issue seems only to occur when the state of the device is on.

## Debug info
Zigbee2MQTT version: 1.17.1-dev commit: 084d6ba (Docker)
Adapter hardware: CC2652R (zzh)
Adapter firmware version: 20210120 (zStack3x0)

Koenkk, 2021-02-12:
I'm not sure if it solves your issue, but did you check https://www.zigbee2mqtt.io/information/groups.html#state-changes ?

crowbarz, 2021-02-17:
Not quite related, the state of the group is always correct, but the _attributes_ of the group keep changing and triggering MQTT updates for the group.

Consider a group `group-1` containing 3 bulbs, each bulb having identical attributes:
```
{ "bulb-1": { "brightness": 50, "state": "OFF" } }
{ "bulb-2": { "brightness": 50, "state": "OFF" } }
{ "bulb-3": { "brightness": 50, "state": "OFF" } }
```
When `group-1` is switched on, one update for the group is received:
```
{ "bulb-1": { "brightness": 50, "state": "ON" } }
{ "group-1": { "brightness": 50, "state": "ON" } // group-1 state changed from OFF to ON
{ "bulb-2": { "brightness": 50, "state": "ON" } } // no group-1 update, group-1 already has "brightness": 50
{ "bulb-3": { "brightness": 50, "state": "ON" } } // no group-1 update, group-1 already has "brightness": 50
```
`bulb-2` and `bulb-3` transitioning to state `ON` don't trigger additional group update messages, because those updates don't change the cached attributes of the group. However, if instead the bulbs have different attributes:
```
{ "bulb-1": { "brightness": 10, "state": "OFF" } }
{ "bulb-2": { "brightness": 20, "state": "OFF" } }
{ "bulb-3": { "brightness": 30, "state": "OFF" } }
```
When `group1` is now switched on, it generates 3 group updates, because the attributes of the group are changed when each of the bulb updates are applied to the group:
```
{ "bulb-1": { "brightness": 10, "state": "ON" } }
{ "group-1": { "brightness": 10, state: "ON" } } // group-1 state changed from OFF to ON
{ "bulb-2": { "brightness": 20, "state": { "old": "OFF", "new": "ON" } } }
{ "group-1": { "brightness": 20, state: "ON" } } // group-1 brightness changed from 10 to 20
{ "bulb-3": { "brightness": 30, "state": { "old": "OFF", "new": "ON" } } }
{ "group-1": { "brightness": 30, state: "ON" } } // group-1 brightness changed from 20 to 30
```
Debouncing group updates prevents intermediate group update messages from being sent by waiting the debounce interval before sending an update for the final state of the group. In the example above, a debounce interval of eg. 1s would wait 1s for all bulb updates to be received and processed before the group update is sent.

Filtering attributes before being applied to a group could reduce the number of group updates sent, as differences in the filtered attributes across group members would not cause group updates to be generated. In the example above, if `brightness` was filtered, then the bulb's `brightness` attribute would not be applied to the group, so only one group update would be sent for the group's transition to state `ON`.

Koenkk, 2021-02-18:
Does the `filtered_attributes` option partially solves your issue? https://www.zigbee2mqtt.io/information/configuration.html#all-devices (can also be applied to groups). 

This should prevent the `brightness ` being published every-time (but you will still get 3 state updates).

crowbarz, 2021-02-18:
Can _all_ of the device specific options be applied to groups? It wasn't clear to me from the documentation that "This [device specific configuration] can also be used in combination with groups" really meant "These options can be used on groups as well as devices".

If so, then there are already _debounce_ and _filtered_attribute_ device specific options. I'll test them out in the next few days and if they work raise a PR on the documentation.

crowbarz, 2021-03-21:
Still need to try to apply device specific options to groups.

closed 2021-04-30 by github-actions[bot]