Koenkk/zigbee2mqtt #6245: Suddenly "port 8485 is already in use" error message. Z2M stopped and not able to start.

FaBRiK74, 2021-02-12:
I use CC2531 (sonoff usb). Worked great until today morning. Nothing was changed in the morning but suddenly Z2M stopped. When I tried to restart it gave an error that the 8485 port is already in use. So I fail to start zigbee2mqtt.

I tried to make rollback with yesteday and day before yesterday full snapshot but got the same error.

I use:
 - Z2M latest version 1.17.1-1
 - HA Core 2021.2.3 (but the problem exixst with 2021.2.1 rolled back)

The log shows this error: https://pastebin.com/VS6AG4uG

Koenkk, 2021-02-12:
I don't think this can solved be solved from z2m. Looks that another application is already using port 8485. 

See https://www.zigbee2mqtt.io/information/frontend.html#frontend how to change the port.

FaBRiK74, 2021-02-12:
Thank you for your answer. It is an interesting issue because we were sleeping while Z2M has been stopped, noone could tousch even a switch. And HA runs on a dedicated RPi 3 (no any other thing is running on it). Also I made a roll back to previous snapshot.
Should I add these two lines to my HA's configuration.yaml? Will it know that it is Z2M's frontend?

frontend:
  port: 8080

Koenkk, 2021-02-13:
Try disabling the frontend entirely by putting `frontend: null` in your config.

FaBRiK74, 2021-02-13:
Finally I wiped the whole HA, installed Z2M, then recovered from the latest snapshot. Now it works great. And I recognised I have to change from CC2531 to a better router.

closed 2021-02-14 by Koenkk