Koenkk/zigbee2mqtt #6247: 2021 New Hue Dimmer Switch is Unsupported

mptr8956, 2021-02-12:
Hello,
i ordered the new hue dimmer switch (that looks slightly different but has the same functions) but it is not recognized by zigbee2mqtt. I tried to follow the instructions for adding a new device, but i am stuck. In the webgui it looks like this:

[image]

[image]

What do I ahve to do to get this thing working?

mptr8956, 2021-02-12:
This is the new dimmer switch i am talking about: https://hueblog.de/2020/12/21/philips-hue-dimmschalter-neue-generation-wird-moderner/

mptr8956, 2021-02-13:
no one who can help/assist?

Koenkk, 2021-02-14:
Device has already been added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2202.

It's supported in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-14 by Koenkk

mabosch74, 2021-02-16:
The device doesn't work. Only the linkquality works.

`info  2021-02-16 10:02:32: Device '0x00178801092699c3' joined
info  2021-02-16 10:02:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00178801092699c3","ieee_address":"0x00178801092699c3"},"type":"device_joined"}'
info  2021-02-16 10:02:32: Starting interview of '0x00178801092699c3'
info  2021-02-16 10:02:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00178801092699c3","ieee_address":"0x00178801092699c3","status":"started"},"type":"device_interview"}'
info  2021-02-16 10:02:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00178801092699c3"},"type":"device_connected"}'
info  2021-02-16 10:02:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00178801092699c3"},"type":"pairing"}'
debug 2021-02-16 10:02:32: Device '0x00178801092699c3' announced itself
info  2021-02-16 10:02:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00178801092699c3","ieee_address":"0x00178801092699c3"},"type":"device_announce"}'
debug 2021-02-16 10:02:33: Device '0x00178801092699c3' announced itself
info  2021-02-16 10:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00178801092699c3","ieee_address":"0x00178801092699c3"},"type":"device_announce"}'
info  2021-02-16 10:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00178801092699c3"},"type":"device_announced"}'
info  2021-02-16 10:02:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00178801092699c3"},"type":"device_announced"}'
debug 2021-02-16 10:02:34: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"modelId":"RWL022"}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:34: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"RWL022"}'
debug 2021-02-16 10:02:35: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Signify Netherlands B.V."}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:35: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"Signify Netherlands B.V."}'
debug 2021-02-16 10:02:35: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:35: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
debug 2021-02-16 10:02:35: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:35: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2}'
debug 2021-02-16 10:02:36: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:36: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":2}'
debug 2021-02-16 10:02:36: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:36: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":1}'
debug 2021-02-16 10:02:36: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:36: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
debug 2021-02-16 10:02:37: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200622"}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:37: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20200622"}'
debug 2021-02-16 10:02:37: Received Zigbee message from '0x00178801092699c3', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.42.0_h1F5E860"}' from endpoint 1 with groupID 0
debug 2021-02-16 10:02:37: No converter available for '324131092621' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"2.42.0_h1F5E860"}'
info  2021-02-16 10:02:37: Successfully interviewed '0x00178801092699c3', device has successfully been paired
info  2021-02-16 10:02:37: Device '0x00178801092699c3' is supported, identified as: Philips Hue dimmer switch (324131092621)
info  2021-02-16 10:02:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Hue dimmer switch","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["on-press","on-hold","on-hold- [... cut ...] "up-press","up-hold","up-hold-release","down-press","down-hold","down-hold-release","off-press","off-hold","off-hold-release"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"324131092621","supports_ota":true,"vendor":"Philips"},"friendly_name":"0x00178801092699c3","ieee_address":"0x00178801092699c3","status":"successful","supported":true},"type":"device_interview"}'
`

ferdyh, 2021-02-17:
I can confirm is still not working. It's added successfully, but unable to use it. It doesn't output any of the actions.

877dev, 2021-02-22:
I have new Hue dimmer v2 and latest dev build, and managed to pair it. I renamed it in configuration.yml to "Hue_dimmer_main_bedroom". 

However I only get this, and nothing sent to my mqtt broker:

```
Zigbee2MQTT:error 2021-02-22 18:37:11: Failed to configure 'Hue_dimmer_main_bedroom', attempt 1 (TypeError: Cannot read property 'write' of undefined

    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices.js:3975:29)

    at async Configure.configure (/app/lib/extension/configure.js:131:13)

    at async Configure.onZigbeeStarted (/app/lib/extension/configure.js:99:17)

    at async Controller.callExtensionMethod (/app/lib/controller.js:385:21))
```

Koenkk, 2021-02-22:
@877dev check https://github.com/Koenkk/zigbee2mqtt/issues/6359#issuecomment-782831408

877dev, 2021-02-22:
@Koenkk I was just looking at theat, however as I use docker-compose the instruction to add new devices are not clear.

I have the files locally in /opt/.... but not sure about the equivalent docker-compose command for:
`docker run -v "$PWD":/app -w /app --rm node:12 npm ci`