Koenkk/zigbee2mqtt #6261: Prolight smart bulb E27 warm white clear (EAN 5412748727432) not supported

johanvh83, 2021-02-13:
I tried the 'how to support new devices' but as I'm running Home Assistant OS and have a basic knowlege, I'm not able to continue.

## Information about the device + link
Device type: bulb E27 8 Watt
Zigbee model: PROLIGHT E27 WARM WHITE CLEAR
Webshop URL: [https://www.hubo.be/nl/p/prolight-smart-classic-led-peerlamp-filament-e27-8w-dimbaar/917710.html](url)
Couldn't found a URL only with the bulb or the manufacturer

## data/database.db entry of the device

`{"id":10,"type":"Router","ieeeAddr":"0xccccccfffe2c5148","nwkAddr":24692,"manufId":4717,"manufName":"Prolight","powerSource":"Mains (single phase)","modelId":"PROLIGHT E27 WARM WHITE CLEAR","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"PROLIGHT E27 WARM WHITE CLEAR","manufacturerName":"Prolight","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190718-532","swBuildId":"2.0"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190718-532","swBuildId":"2.0","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1613232473298}
`

Koenkk, 2021-02-14:
Added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-14 by Koenkk

johanvh83, 2021-02-14:
Hi, many thanks to add this device. It's supported in the Edge version and works fine.
Only the picture is the default Zigbee logo and not a picture of the bulb.
When I may suspect it will be integrated in the 'normal' Zigbee2MQTT add-on?

Koenkk, 2021-02-14:
Picture will be available once its in a release, which will be on 1 march (next release)