Koenkk/zigbee2mqtt #6262: Color Temperature Correction\Calibration

RynoL37, 2021-02-13:
## Is your feature request related to a problem? Please describe
Depending on the color temperature of LEDs used in WW\CW LED strips, a specific color temperature\ mired value sent to the LED controller will have different color temperatures. Example, sending 460 mireds to one LED controller results in 2700K which is a mired of 370.

## Describe the solution you'd like
Matching other lights will require a correction map similar to hue correction mapping for RGB. Might only be necessary for LED controllers where various WW\CW combinations might be connected.

## Describe alternatives you've considered
None

## Additional context

sjorge, 2021-02-14:
Please enable debug logging and check if you're not hitting the clamping to min/max here:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/bbcf519385c213e22efa2ed29cce5519dd1b29a5/converters/toZigbee.js#L718

https://github.com/Koenkk/zigbee-herdsman-converters/blob/bbcf519385c213e22efa2ed29cce5519dd1b29a5/lib/light.js#L37-L47

RynoL37, 2021-02-14:
Correct, my 'reference' bulbs only reach 370 at its warmest. However, the LED controllers I am using, Gledopto GL-C-008P, are not clamping until much later and produce visible changes across the mired range. I'm assuming since they are analog controllers and do not know the exact color temp of the warm and cold LEDs connected , that the mired range sent to them is basically mixing full warm to full cold leds.

I measured the color temperature, and while there are measurable differences between specific bulbs its not visibly noticeable, however the LED Controllers for my LED strips are significantly and very noticeably different.

I did run the debug log and you are correct, I saw the clamping messages for both my 'reference' bulb and my gledopto LED controller. The Gledopto clamps at 495 however which explains my noticeable changes up near that range.

`Zigbee2MQTT:debug 2021-02-14 07:13:34: Requested color_temp 399 is higher than maximum supported 370, using maximum!`
`Zigbee2MQTT:debug 2021-02-14 07:15:33: Requested color_temp 500 is higher than maximum supported 495, using maximum!`

sjorge, 2021-02-14:
Oh so you can just use whatever random cold/warm led strip with these controllers? Interesitng... that would indeed require calibration of sort.

I think it needs to be applied before the clamping code. So when setting 400 results in actual 370 we still need to settable value withing range. Of course the reverse would need to be applied too to the read value.

closed 2021-03-24 by github-actions[bot]

oldcrazyeye, 2021-07-02:
Would be interested in color temp calibration as well. I have 20+ ecosmart bulbs that max out at 370 mired. It would be nice to have the slider only go up to that value vs. 500mired. Also when you adjust the color temp in HA it will spring back to 370 again vs 500. Just need some slider adjustment/calibration if possible.

RynoL37, 2021-07-02:
> Would be interested in color temp calibration as well. I have 20+ ecosmart bulbs that max out at 370 mired. It would be nice to have the slider only go up to that value vs. 500mired. Also when you [...]

That isn’t what was proposed discussed. The base of my issue was that 300 mired on my leds was not outputting the correct color temperature and a mapping is needed to get the expected output for a given input. The lights don’t match other bulbs.

 What you are looking for is a UI change to not show values outside the bulbs capability.

sjorge, 2021-07-02:
The info for those bulbs is probably missing

https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

Once those are added ha should be aware of the correct min and max values.
~ sjorge

oldcrazyeye, 2021-07-02:
Yea my bad for the confusion

thatslolo, 2023-07-03:
Did this go anywhere? I have the same problem that my lightstrip on the gledopto controller has is only going to 333 mireds, but the controller exposes the possibility to set it to 495 if I understand the converter correctly.
https://github.com/Koenkk/zigbee-herdsman-converters/blob/4a8ab6fb8c19ba30d3596321b8a3fa2e5661036c/src/devices/gledopto.js#L291C74-L291C74