Koenkk/zigbee2mqtt #6266: many clicks push were loss with domoticz in the Xiaomi WXKG01LM (zigbee2mqtt definition/exposes missing)

kitopopo, 2021-02-14:
Dear @koenkk,

Due to the change of  domoticz plugin from 0.2.1 version to 3.0 version we have lost the many clicks function in domoticz with the xiaomi switch (WXKG01LM). This type of clicks is very useful and is detected by domoticz as "many" but the error appears in domoticz:

`(zigbe2mqtt) Unable to find selector switch level for value "many"`

this is the zigbe2mqtt log:

`feb 10 23:02:15 raspberrypi npm[486]: Zigbee2MQTT:info 2021-02-10 23:02:15: MQTT publish: topic 'zigbee2mqtt/Switch mesita', payload '{"action":"many","battery":100,"linkquality":76,"voltage":3072}'
`

I have been asking @stas-demydiuk  to reintroduce the click type "many click" again in the new version of the plugin but he comments that the problem is in the definition of zigbee2mqtt, I attach the zigbe2mqtt definition of the xiaomi switch:

```
{
   "definition":{
      "description":"MiJia wireless switch",
      "exposes":[
         {
            "access":1,
            "description":"Remaining battery in %",
            "name":"battery",
            "property":"battery",
            "type":"numeric",
            "unit":"%",
            "value_max":100,
            "value_min":0
         },
         {
            "access":1,
            "description":"Triggered action (e.g. a button click)",
            "name":"action",
            "property":"action",
            "type":"enum",
            "values":[
               "single",
               "double",
               "triple",
               "quadruple",
               "hold",
               "release"
            ]
         },
         {
            "access":1,
            "description":"Link quality (signal strength)",
            "name":"linkquality",
            "property":"linkquality",
            "type":"numeric",
            "unit":"lqi",
            "value_max":255,
            "value_min":0
         }
      ],
      "model":"WXKG01LM",
      "vendor":"Xiaomi"
   },
   "endpoints":{
      "1":{
         "bindings":[
            
         ],
         "clusters":{
            "input":[
               
            ],
            "output":[
               
            ]
         },
         "configured_reportings":[
            
         ]
      }
   },
   "friendly_name":"Switch salon",
   "ieee_address":"0x00158d000418cd86",
   "interview_completed":true,
   "interviewing":false,
   "model_id":"lumi.sensor_switch",
   "network_address":24485,
   "power_source":"Battery",
   "supported":true,
   "type":"EndDevice"
}

```

As you can see in the definition, the "many click" type of click with respect to previous versions has disappeared in the "values" field.

this is the picture from v.2.1 plugin

[image]

this is the picture from v.2.1 plugin

[image]

Could you include this type of click again in the definition of the xiaomi switch? Then I attach the request to @stas-demydiuk :

https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/572

I wait news from you,

Best regards.-

Koenkk, 2021-02-14:
Added back `many`. 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-14 by Koenkk

kitopopo, 2021-02-14:
> Added back `many`.
> 
> Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

thanks. You are the best.  thanks a lot.

Best regards