Koenkk/zigbee2mqtt #6267: TS0601_cover: state problem

Hacker-CB, 2021-02-14:
I have some of Tuya cover motor. 

Problem is that it report `2` instead of `0` when is fully closed, so `CLOSED` state is never published and HA `cover.toggle` service does now work.
Also Jagger from https://t.me/zigbeer has same motor and he told me that his motor reports `97` insetad of `100` when is fully opened.

Current code for state: (https://github.com/Koenkk/zigbee-herdsman-converters/blob/bbcf519385c213e22efa2ed29cce5519dd1b29a5/converters/fromZigbee.js#L1506)

```javascript
if (position > 0 && position <= 100) {
    return {running: false, position: position, state: 'OPEN'};
} else if (position == 0) { // Report fully closed
    return {running: false, position: position, state: 'CLOSE'};
} else {
    return {running: false}; // Not calibrated yet, no position is available
}
```

So, I suggest to rewrite this to:
```javascript
if (position > 50 && position <= 100) {
	return {running: false, position: position, state: 'OPEN'};
} else if (position >=0 && position <= 50) { // Report fully closed
	return {running: false, position: position, state: 'CLOSE'};
} else {
        return {running: false}; // Not calibrated yet, no position is available
}
```

@Koenkk please let me know if its ok and I will prepare PR.

**My MQTT state when closed:** (state is not `CLOSED`)
[image]

**My MQTT state when opened:** (state is ok)
[image]

Koenkk, 2021-02-14:
@DomiStyle are you fine with this?

DomiStyle, 2021-02-14:
@Koenkk Maybe we can add a calibration parameter which defaults to 0 and can be changed by people where strange positions are returned? So not everyone is forced to > 50 by default.

However, Home Assistant might still not recognize the close/open state properly since they go by position I think.

Hacker-CB, 2021-02-15:
@DomiStyle Toggle in HA will work normally if we will detect state based on `50`. In any case it will work

DomiStyle, 2021-02-15:
@Hacker-CB That's not what I meant.

The state attribute is not used for the Home Assistant integration, only the position. If we report `{running: false, position: 2, state: 'CLOSE'}` Home Assistant will still think the cover is open and not trigger any automation tied to the curtain being closed.

To test this you can publish these MQTT messages to your motor topic:

`{"linkquality":105,"position":100,"running":false,"state":"OPEN"}`

`{"linkquality":105,"position":2,"running":false,"state":"CLOSE"}`

Home Assistant will not recognize that the cover has been closed. The correct implementation would be to add a calibration parameter and pull all values under it to 0. That way it works with HA too and doesn't break anything.

Hacker-CB, 2021-02-15:
@DomiStyle according HA sources `toggle` depends on state, not on position:

[components/cover/__init__.py](https://github.com/home-assistant/core/blob/c5b9ad83c2a8f363a3f912f542ce966034ee1274/homeassistant/components/cover/__init__.py#L261)

```python
def toggle(self, **kwargs: Any) -> None:
    """Toggle the entity."""
    if self.is_closed:
        self.open_cover(**kwargs)
    else:
        self.close_cover(**kwargs)
```

[components/mqtt/cover.py](https://github.com/home-assistant/core/blob/c5b9ad83c2a8f363a3f912f542ce966034ee1274/homeassistant/components/mqtt/cover.py#L406)

```python
def is_closed(self):
    """Return true if the cover is closed or None if the status is unknown."""
    if self._state is None:
        return None

    return self._state == STATE_CLOSED
```

Also HA has 4 states for MQTT cover, but current tuya implementation in z2m uses only 2 of them:

```python
STATE_OPEN = "open"
STATE_OPENING = "opening"
STATE_CLOSED = "closed"
STATE_CLOSING = "closing"
```

DomiStyle, 2021-02-15:
@Hacker-CB Home Assistant supports it, the Z2M integration doesn't use it. There is no `state_topic` set by Z2M for the MQTT discovery to use:

```
{
  "availability": [
    {
      "topic": "zigbee/bridge/state"
    }
  ],
  "command_topic": "zigbee/livingroom_curtain_main/set",
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x00"
    ],
    "manufacturer": "TuYa",
    "model": "Curtain motor/roller blind motor/window pusher/tubular motor (TS0601_cover)",
    "name": "livingroom_curtain_main",
    "sw_version": "Zigbee2MQTT 1.17.1"
  },
  "json_attributes_topic": "zigbee/livingroom_curtain_main",
  "name": "livingroom_curtain_main",
  "position_topic": "zigbee/livingroom_curtain_main",
  "set_position_template": "{ \"position\": {{ position }} }",
  "set_position_topic": "zigbee/livingroom_curtain_main/set",
  "unique_id": "0x086bd7fffedae046_cover_zigbee",
  "value_template": "{{ value_json.position }}"
}
```

Even if, the state strings do not match (CLOSE vs closed, OPEN vs open) and would need to be set by Z2M too (state_open,  state_closed, ...), which is probably a bigger change that's needed at the Z2M core.

Hacker-CB, 2021-02-15:
@DomiStyle you are right, seems `cover.toggle` service in HA should work after `state_topic` will be set.

 I think it will be good to add `withState()` feature to curtain exposes. @Koenkk ?

Koenkk, 2021-02-15:
> I think it will be good to add withState() feature to curtain exposes. @Koenkk ?

Agree, it's missing now indeed. Could you make a PR?

Hacker-CB, 2021-02-15:
@Koenkk, ok, but a little bit later, very busy now. I hope I will find time till end of this week.

closed 2021-03-26 by github-actions[bot]

MacieIKS, 2021-09-18:
@Hacker-CB
Refreshing this topic, I noticed similar problem, all my MOES AM43 (TS0601) are reporting most often CLOSE with position:1. Perhaps, did you find any workaround for that?

Thanks you.

luukdobber, 2021-12-07:
I'm running into this problem with my Tuya TS0601 curtain motor as well. It often reports 1 or 2 instead of 0 and 97 instead of 100. A calibration option to tweak the open/close threshold could solve the incorrect state.

Could we reopen this issue @Koenkk ? Perhaps I can give it a try to implement the state_topic and calibration option, but my typescript skills are a bit rusty 😛

Surimisalade, 2023-04-06:
Has this been updated yet? I am also running in trouble with the same kind of behaviour. The curtain is fully open but returns a value of 90 instead of 100.
Would be great to be able to change it, so that 90 gets displayed as 100 and the curtain state is open. Also the open button gets grayed out then, now i can still open it, even if it is fully open.

henrykuijpers, 2024-03-27:
@Koenkk any update / news on this? I would be willing to create a PR for this, but it would be nice to receive a few pointers on what to do. 

I do think calibration is the way to go, as also my covers run until 1/2/3% and thus don't go to 0. 
Also, I usually see 97/98/99 instead of 100%.

Would be good to be able to tune these settings and then make the state go to closed. 

Btw, I also see sometimes, when the cover is closed, that the motor decides "hey, I need to go back a bit" and then creates a 10cm opening again. :P Probably it "felt" that the curtain was providing some resistance. Something I shouldn't fix with software I guess. :P