Koenkk/zigbee2mqtt #6272: Zigbee2mqtt with RBCC2652RB stops working every 7-8 days and need to be re-flashed. Zigbee buttons ruined after recovery

mcividino, 2021-02-14:
Zigbee2mqtt stops working ever 7 to 8 days. Last weekend I had the same issue and has to spend the whole weekend reinstalling all the software back in the Raspberry Pi 4 (I use IOTstack) and re-flashing the coordinator.

When the failure happens, I can see the MQTT messags published by the buttons, but it fails to publish the ON/OFF messages to the bulbs.
Next I stopped the Docker contaoiners and rebooted the server.
Aftert the server is online now there is absolutely no communication between zigbee2mqtt and any zigbee devices.

In the mqtt debug messages I can see in Node Red, I constantly get these:
14/2/2021 12:40:50node: zigbee2mqtt/bridge/loggingzigbee2mqtt/bridge/logging : msg.payload : string[69]
"{"level":"error","message":"Failed to execute LQI for 'Extender-01'"}"
14/2/2021 12:41:01node: zigbee2mqtt/bridge/loggingzigbee2mqtt/bridge/logging : msg.payload : string[74]
"{"level":"error","message":"Failed to execute LQI for 'Lamp_Est_Mariano'"}"
14/2/2021 12:41:10node: zigbee2mqtt/bridge/loggingzigbee2mqtt/bridge/logging : msg.payload : string[70]
"{"level":"error","message":"Failed to execute LQI for 'Luz_Escalera'"}"
14/2/2021 12:41:20node: zigbee2mqtt/bridge/logging
zigbee2mqtt/bridge/logging : msg.payload : string[69]
"{"level":"error","message":"Failed to execute LQI for 'Luz_Salon_2'"}"

Today  had the same issue and we cannot control the lights at home anymore. I re-flashed the dongle and there seems to be communication again with the devices.
After flashing I had to re-pair all the devices.

All devices seem to work, except all my Aqara OPPLE buttoms which seem to be ruined. They are binded with every single other zignbee devoice at home and they turn everything on and off. This happens now even if zigee2mqtt is not running.

Last week I had the same issue with three buttons and I had to replace them. Now all of them seem to be ruined the same way, So I have to buy all new buttons.
Last wek I asked how to fix these three buttons I lost, but nobody seem to know how to fix them. They seem to be permanently like that.
Now it will cost memore than 200 euros to buy all new ones and will be without switches for several weeks.

These are all the version details, as well as configuration file:
- Docker Version:
pi@raspberrypi:~/IOTstack/volumes/zigbee2mqtt/data $ docker version
Client: Docker Engine - Community
 Version:           20.10.1
 API version:       1.41
 Go version:        go1.13.15
 Git commit:        831ebea
 Built:             Tue Dec 15 04:35:27 2020
 OS/Arch:           linux/arm
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.1
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       f001486
  Built:            Tue Dec 15 04:33:02 2020
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.4.3
  GitCommit:        269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc:
  Version:          1.0.0-rc92
  GitCommit:        ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
pi@raspberrypi:~

- Zigbee2mqtt versions:
# cat package.json
{
  "name": "zigbee2mqtt",
  "version": "1.17.0",
  "description": "Zigbee to MQTT bridge using Zigbee-herdsman",
  "main": "index.js",
  "repository": {
    "type": "git",
    "url": "git+https://github.com/Koenkk/zigbee2mqtt.git"
  },
[... 272 lines cut ...]
    friendly_name: Luces_Terraza
    devices:
      - 0x00124b0021928a7d/11
  '11':
    friendly_name: Luces_Lola
    devices:
      - 0xf0d1b80000129241/1
      - 0xf0d1b8000012ba0d/1
      - 0xf0d1b8000012c249/1
      - 0xf0d1b8000012caac/1
  '12':
    friendly_name: Luces_Anna
    devices:
      - 0xf0d1b8000014a4fd/1
      - 0xf0d1b8000014a3a9/1
      - 0xf0d1b8000014a3bf/1

- Coordinator:
CC2652RB - from Slaeh
Version - CC2652RB_20201026.hex

- zigbee2mqtt/bridge/config MQTT message:

14/2/2021 12:51:51node: zigbee2mqtt/#zigbee2mqtt/bridge/config : msg.payload : string[284]
"{"commit":"07cdc9d","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":6760},"permit_join":true,"version":"1.17.0"}"

Koenkk, 2021-02-14:
- Try updating to the latest firmware (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin), fixes some stability issues
- Don't do the network scan on a regular base but only when you need it . It is very heavy for the network: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestnetworkmap

mcividino, 2021-02-14:
Thanks. Will try a new firmware.
Do you kow any way to unbind thebuttons that have rendered useless. They seem to be bind with every other zigbe device in the home.
How do I know to which devices they are bind?

mcividino, 2021-02-14:
I anser to myself.
Whe issue observed was similar to this one:
https://github.com/Koenkk/zigbee2mqtt/issues/3926

I found the way to unbind the buttons:
1) Force delete the device
2) Stop Zigbee2MQTT
3) press the pairing button on the device for 5 seconds. After this the device will unboud.
4) start Zigbee2MQTT
4) Pair the device again to zigbee2mqtt using the pairing button. HOWEVER, **do not use** the same friendly name or the issue will happen again. 

I updated the coordinator with the latest FW as indicated. Let's see how it behaves in a few days.

Thanks for your help.

closed 2021-03-26 by github-actions[bot]